espressif/esp-at #156: SDIO AT Command

sathishgkg, 2019-01-14:
Hi, 
IS there any pin configuration for sdio at communication between master and salve.

xcguang, 2019-01-14:
Hi,
Slave configuration:
[image]

master pin is the same as slave.

https://espressif-docs.readthedocs-hosted.com/projects/esp-idf/en/latest/api-reference/peripherals/sd_pullup_requirements.html maybe is helpful for you.

We test the project on our internal test board, the schematic is 
[SCH_ESP-SDIO-TESTBOARD_V1.pdf](https://github.com/espressif/esp32-at/files/2754095/SCH_ESP-SDIO-TESTBOARD_V1.pdf)

sathishgkg, 2019-01-14:
[image]
im not able to see the window which you mentioned above

sathishgkg, 2019-01-18:
Hi,
After successful flash at command firmware for sdio: in make monitor I'm getting this error
E (760) SDIO-AT: Cannot get read data address.

sathishgkg, 2019-01-20:
Hi, 
im trying to interface esp32(At command - SDIO) with rx65n using sdio protocol, in monitor it print like  E (760) SDIO-AT: Cannot get read data address.  how to solve this issue?

Thanks
Regards
Sathish

xcguang, 2019-01-20:
Hi,
Could you read the data, when master sends data?
Maybe this log level should not be ESP_LOGE in `at_sdio_read_data `, we will change it to ESP_LOGI.
If you always can not get the data, can you capture the SDIO bus raw data?

ESP-Coco, 2019-06-18:
Can you provide the corresponding log for us to analyze?

closed 2019-11-06 by xcguang