espressif/esp-at #159: Error code 0x01030000 from command "AT"

maxi03462, 2019-01-15:
Hi,
I am testing the AT commands, by repeatedly sending the command "AT\r\n" I receive the error code:

```
AT
ERR CODE:0x01030000

ERROR
```

As long as the same command is still sent, the same error is still received.

When sending any other command after receiving this error, the same fault message is received again, regardless of the command that is sent.

Only by sending a different command twice than the one that generated the error, the module responds correctly.

I attach the "sdkconfig"
[sdkconfig.zip](https://github.com/espressif/esp32-at/files/2759758/sdkconfig.zip)

xiewenxiang, 2019-01-16:
Please provide the version you are using now
`AT+GMR`

maxi03462, 2019-01-16:
This is the response of the command:

```
AT+GMR
AT version:1.3.0.0-dev(6aafec8 - Dec  7 2018 02:23:19)
SDK version:v3.3-dev-343-g7fa98593b
compile time:Jan 15 2019 16:57:32

OK
```

xiewenxiang, 2019-01-22:
I tested it, but I failed to reproduce your issues. Did you compile the firmware by yourself? Are there any changes other than sdkconfig?

maxi03462, 2019-01-23:
Download and compile as indicated in the readme file.
The sdkconfig file has the classic bluetooth activated as well as the AT commands for the classic bluetooth.

Try to compile with the default options and the problem persists.

The error appears when the same command is sent several times, waiting approximately 1 second from "OK" to the command sent.

The command "xtensa-esp32-elf-gcc -v" returns:

```
Using built-in specs.
COLLECT_GCC=xtensa-esp32-elf-gcc
COLLECT_LTO_WRAPPER=/home/electro2/esp/xtensa-esp32-elf/bin/../libexec/gcc/xtensa-esp32-elf/5.2.0/lto-wrapper
Target: xtensa-esp32-elf
Configured with: /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/configure --build=x86_64-build_pc-linux-gnu --host=x86_64-build_pc-linux-gnu --target=xtensa-esp32-elf --prefix=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf --with-local-prefix=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf/sysroot --with-sysroot=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf/sysroot --with-newlib --enable-threads=no --disable-shared --with-pkgversion='crosstool-NG crosstool-ng-1.22.0-80-g6c4433a' --disable-__cxa_atexit --enable-cxx-flags='-fno-rtti -ffunction-sections' --with-gmp=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-mpfr=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-mpc=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-isl=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-cloog=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-libelf=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --enable-lto --enable-target-optspace --without-long-double-128 --disable-libgomp --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libquadmath-support --disable-nls --enable-languages=c,c++ --disable-libstdcxx-verbose --enable-threads=posix --enable-gcov-custom-rtio
Thread model: posix
gcc version 5.2.0 (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 
```

wnoliveira, 2019-04-25:
I have the same problem with the pre-compiled binary available in the official website. 

I've tracked the moment were the error begins. In my case, seems like if you send data through the serial port while the module is sending you a received message (started with "+IPD") it just crashes the AT firmware flow.

The big problem is that, after the first error occurs, the firmware does not work to a long time (something as 1 or 2 minutes). It just returns the same error code (0x01030000) as response for all commands.

This seems a big problem...

swouf, 2019-07-31:
Hello, I can confirm I encounter the same problem. Documentation on my case will follow.

xiewenxiang, 2019-08-01:
Did you have configured the char delay in the serial tool?
If so, please turn this configuration off.

closed 2019-11-06 by xcguang