espressif/esp-at #161: AT+BLEGATTCWR execution problem

kmakhil90, 2019-01-17:


xiewenxiang, 2019-01-21:
If use ESP32 AT Client write data to the characteristic, The data will be written exactly as it was written.

If use ESP32 AT Client write data to the descriptor(UUID: 0x2902), the data inputed by the user will be ignored. If the characteristic is notification, will write 0x01 to the  remote device, if the characteristic is indication, will write 0x02 to the remote device.

kmakhil90, 2019-01-21:
Hi,
 Now I added one more characteristics with read and write property and tried to write some data by using AT+BLEGATTCWR=0,3,2,,3 (3 bytes) from TERMITE terminal. But after getting OK message, i have  read the same characteristics and not showing  the valid data here. it means the command not writing the data to the characteristic.
Any suggestions to solve the problem?

xiewenxiang, 2019-01-22:
Do you mean that the client writes data to the server, but the server doesn't have a log?

kmakhil90, 2019-01-23:
> Do you mean that the client writes data to the server, but the server doesn't have a log?

Yes, When i write from an ESP32 client not getting any indication in esp32 server as log. But if i try with nRF app as client it shows a WRITE indication in esp32 server log and written the data. 
(The snapshots attached above in this discussion).

kmakhil90, 2019-01-23:
> But I saw the prompt in your screenshot....
> 
> `+WRITE:<conn_index>,<service_index>,<char_index>,<desc_index>,<length>,<data>`

This message received when we write data from nRF app (Client) and not from ESP32.

xiewenxiang, 2019-01-23:
The first image should be ESP32 client writing data to server, write ccc is still correct, but write char don't have log. I think it is because the data length is big than attribute value.

1.  which version of AT do you used?
2. Do you using the default service bin? If not, please provide the service bin which you are used.
3. Please try `AT+BLEGATTCWR=0,3,1,,2`

d-kato, 2019-05-02:
@xiewenxiang, I have the same problem. I am using "ESP32 AT Bin V1.1.3".
It seems that "AT+BLEGATTCWR" does not work properly when the property of the characteristic is **"read-write (0x0a)"**.
```
AT+BLEGATTCPRIMSRV=0
+BLEGATTCPRIMSRV:0,1,0x1801,1
+BLEGATTCPRIMSRV:0,2,0x1800,1
+BLEGATTCPRIMSRV:0,3,0xA000,1

OK
AT+BLEGATTCCHAR=0,3
+BLEGATTCCHAR:"char",0,3,1,0xA001,0x0a
```

If the property of the characteristic is **"write (0x08)"**, it seems to work properly.
```
AT+BLEGATTCPRIMSRV=0
+BLEGATTCPRIMSRV:0,1,0x1801,1
+BLEGATTCPRIMSRV:0,2,0x1800,1
+BLEGATTCPRIMSRV:0,3,0xA000,1

OK
AT+BLEGATTCCHAR=0,3
+BLEGATTCCHAR:"char",0,3,1,0xA001,0x08
```

xcguang, 2020-03-15:
Can you try the latest master branch?

xiewenxiang, 2020-03-19:
@d-kato 
If you still have similar problems, please provide your custom service table and reopen an issue.

xiewenxiang, 2020-03-19:
Please use the latest master branch or release/v2.0

closed 2020-03-19 by xiewenxiang