espressif/esp-at #164: bluetooth classic with AT commands

erdemp, 2019-01-19:
Hi guys,
I have ESP-WROOM-32 module and I try to make BT connection with PC or a mobile application. I have checked post and documents. I found some sample codes for this and I used them like below.
```
OK
AT+BTINIT=1
OK
AT+BTSPPINIT=2
OK
AT+BTSPPSTART
OK
AT+BTNAME?
+BTNAME:ERDEMBT
OK
AT+BTSCANMODE=2
OK
AT+BTSECPARAM=3,1,"1234"
OK
+BTSPPCONN:0,e4:b3:18:5b:36:54
AT+BTSPPSEND
OK
>modul modul
pc pc pc pc pc  
```
This code works fine but I want to make connection with pin code. I dont want anybody to connect my device. After above code  pc or mobile app. just scan and connect to ESP32 and start to send datas.  I want to define a pin code in to my ESP32 module.  Other device ( pc , mobile application or any other BT device) should ask user for this code. If user knows ESP32's pin then it can make a connection. 

Do you have any idea how can I do this with AT commands?

igrr, 2019-01-19:
Looks like a duplicate of #162?

xiewenxiang, 2019-01-21:
ESP32 supports both Simple pair and Legacy pair by default. If the remote device support Simple pair, they will use the Simple pair encryption, the PIN code will be ignored.

If you must input the PIN code, you can disable Simple pair in ESP32 side. 
You need to delete the following lines of code and recompile the project

PATH: `esp-idf/components/bt/bluedroid/device/controller.c`

```c
simple_pairing_supported = HCI_SIMPLE_PAIRING_SUPPORTED(features_classic[0].as_array);
if (simple_pairing_supported) {
    response = AWAIT_COMMAND(packet_factory>make_write_simple_pairing_mode(HCI_SP_MODE_ENABLED));
    packet_parser->parse_generic_command_complete(response);
}
```

closed 2020-01-09 by xcguang