espressif/esp-at #169: Documentation GATT Service Bin

HyP3r-, 2019-02-06:
The documentation on how to create a GATT Service CSV is not really accurate and assumes automatisms that are not defined in the standard.

In the first step it should be explained that GATT structures always consist of [services](https://www.bluetooth.com/specifications/gatt/services), [declarations](https://www.bluetooth.com/specifications/gatt/declarations) and optional [descriptors](https://www.bluetooth.com/specifications/gatt/descriptors).

Thus, for example, the first entry should always be a [primary service](https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.attribute.gatt.primary_service_declaration.xml) (UUID 0x2800) which determines with its value which service is described (for example, a predefined one such as [0x180A](https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.service.device_information.xml) or a self generated one).

Services always consist of Characteristic Declarations which have as UUID [0x2803](https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.attribute.gatt.characteristic_declaration.xml) and the Value describes which read and write rights are available. The standard describes that the other bytes define the Value Handle and Characteristic UUID, but should not be appended to this CSV (why?).

As a further entry follows the actual attribute. The value of the attribute contains the value which can be written or read by a foreign device. It is important that the Characteristic Declaration authorizations for the read/write authorizations of this attribute match.

The attribute can be described further by descriptors. A special one is the descriptor "[Client Characteristic Configuration](https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.descriptor.gatt.client_characteristic_configuration.xml)" (UUID 0x2902) which should be present if the Notify bit has been activated in the Characteristic Declaration (UUID 0x2803). This descriptor should then always be writable and readable.

tehKaiN, 2019-07-02:
I'm struggling to add custom GATT service partition. [Current documentation](https://github.com/espressif/esp32-at/blob/master/tools/README.md) mentions three steps:

> 1. define the service in xls or csv file
> 1. use BLEService.py to generate
> 1. download generated bin to address defined in partition table

but only steps 1 and 2 are further described. How to do the third one?

HyP3r-, 2019-07-02:
>     * What is the `address defined in partition table`? Is it `at_customize` defined in partitions_at.csv? I think not, because from what I can see it loads partitions defined in at_customize.csv.
When you build this project there will be a lot of ***.bin** files generated and one ***.bin** files is the **ble_data.bin**. You should find it in `build/customized_partitions`. This file has to be written in flash on the correct offset. But when you do `make print_flash_cmd` should see how and where you have to write this file. The offset is technically that offset which you configure int the `at_customize.csv`.

> Also, I think it would be helpful to have GATT service example for BLE SPP mode, because creating GATT partitions is needed in "UART-BLE Passthrough Mode" example chapter in documentation.
It seems like the [example.csv](https://github.com/espressif/esp32-at/tree/master/components/customized_partitions/raw_data/ble_data) is the nessercary GATT Configuration for using the BLE SPP Feature because the listed Characteristic Declarations in the [Manual](https://www.espressif.com/sites/default/files/documentation/esp32_at_instruction_set_and_examples_en.pdf) in the chapter 9.5.2.2 are the same. But I'm also not sure.