espressif/esp-at #172: Filter BLE address while scanning

carlessole, 2019-02-08:
Hi everybody,
I'm doing a BLE scan using AT cmds (v1.1.3.0). The procedure I follow is:
```
AT+BLEINIT=1  //client role
AT+BLESCANPARAM=1,0,0,100,50   //Set scan param
AT+BLESCAN=1   //Start scanning until AT+BLESCAN=0 is sent
```
Everything works fine but the problem is that so many devices are found, and then, there is a lot of trafffic in AT Uart.
So my question is:

Is there a way to filter the scanning and just scan for the BLE address that starts with some pattern or something like that?

Thank you very much.

Rvice, 2019-02-08:
The documentation for the AT commands looks like there is some form of filtering, but it was never finished or implemented. Currently, we just filter for specific MAC addresses or for a specific data packet. Hopefully, there is something that can be done inside the ESP32 to perform this filtering though.

xiewenxiang, 2019-02-12:
Filtering can be configured in menuconfig. Is this configuration suitable for your application?

`make menuconfig->Component config->Bluetooth->Bluetooth controller`

carlessole, 2019-02-12:
Thank @xiewenxiang ,
But I don't see 
`make menuconfig->Component config->Bluetooth->Bluetooth controller`
in my project.
What I see is:
[image]

But in 
 `make menuconfig->Component config->Bluetooth->Bluedroid Enable`
I have these options:
[image]
So maybe are you referring to these three final fields?

Thank you.

carlessole, 2019-02-25:
So, I understant that there is no way using AT CMDS to filter the advertisment packets during scan?

Thanks again

wibbler, 2022-01-29:
It would be very, very useful to be able to set the 'scan duplicate' element  in the scan_params structure in the esp_err_t esp_ble_gap_set_scan_params(esp_ble_scan_params_t *scan_params) function call. Without this the serial port and/or cpu may become overwhelmed when scanning with many BLE devices available.  It is suggested that this is implemented by adding another parameter to the BLESCANPARAM.

xcguang, 2022-01-30:
`AT+BLESCAN=<enable>[,<interval>][,<filter_type>,<filter_param>]`

You can try the last two parameters

wibbler, 2022-01-31:
Thanks for that, I had seen this and tested it. The problem is when you do not know either the device MAC nor Bluetooth address (or perhaps using a random address) but wish to scan for a device that supports a particular service (such as a heart monitor) in a busy environment. At present this means having to needlessly decode hundreds (or tens thousands) of duplicate advertising reports when scanning. This obviously eats into the serial port bandwidth and available CPU. If this is not seen as important I guess I can compile a custom command to drop the duplicates when starting BLE scanning.