espressif/esp-at #173: Add BLE callback keeping the original one

carlessole, 2019-02-08:
Hello,
I'm trying to add a callback for BLE events using the code origingally from the release v1.1.3.0.
To do that I just:
`esp_err_t res =  esp_ble_gap_register_callback(my_ble_eventHandler);`

The callback is setted ok, bu the problem is when I do so, I overwritte the original callback that libat_core.a sets and then all AT+CMDS from AT does not work properly anymore.
Is there any way to set custom callback while maintaing the original way set by libat_core.a?

Thank in advance.

xiewenxiang, 2019-02-12:
This is not yet supported.

carlessole, 2019-02-25:
@xiewenxiang thanks.
And do you know if in coming releases this will be supported?
Thanks.

carlessole, 2019-02-27:
Hi everybody.
Finally I have managed the way to keep the old callback.
The solution here:
```
#include <stdio.h>
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "esp_log.h"
#include "esp_bt.h"
#include "esp_gap_ble_api.h"
#include "btc_manage.h"

static esp_gap_ble_cb_t old_ble_gap_callback = NULL;

static void custom_ble_gap_callback(esp_gap_ble_cb_event_t event,	esp_ble_gap_cb_param_t *param)
{
        //add your custom functionallity here

	if(old_ble_gap_callback != NULL)
	{
		 old_ble_gap_callback(event, param);
	}
}

bool custom_set_ble_gap_register_callback(void)
{
	bool ret = false;
	if(old_ble_gap_callback == NULL)
	{
		ESP_LOGD(_TAG, "%s: Registering custom BLE GAP callback", __FUNCTION__);
		//First we get the current callback
		old_ble_gap_callback = (esp_gap_ble_cb_t)btc_profile_cb_get(BTC_PID_GAP_BLE);
		if(old_ble_gap_callback == NULL)
		{
			ESP_LOGD(_TAG, "%s: No existing BLE GAP callback...", __FUNCTION__);
		}
	}

	esp_err_t res = esp_ble_gap_register_callback(custom_ble_gap_callback);
	if(res == ESP_OK)
	{
		ret = true;
		ESP_LOGD(_TAG, "%s: Custom BLE GAP callback registered successfully!", __FUNCTION__);
	}
	else
	{
		ESP_LOGD(_TAG, "%s: Unable to register custom BLE GAP callback...Error num: %d", __FUNCTION__, res);
	}

	return ret;
}
```

closed 2019-02-28 by carlessole