espressif/esp-at #174: AT+BLEGATTCPRIMSRV=0 crashes and makes ESP32 reboot

carlessole, 2019-02-11:
Hi.
I have a problem using the latest version v.1.1.3.0.
ESP32 is acting as client and after a succesful connection with this device: [SensorTAG CC650](http://www.ti.com/tool/cc2650stk) if I try to discover all its services ESP32 reboots.
I send from AT
```
[09:20:13:058] AT+BLEINIT=1␍␊
[09:20:13:570] ␍␊
[09:20:13:570] OK␍␊
[09:20:16:946] AT+BLECONN=0,"54:6c:0e:4d:53:03"␍␊
[09:20:16:946] ␍␊
[09:20:16:946] OK␍␊
[09:20:17:154] +BLECONN:0,"54:6c:0e:4d:53:03"␍␊
[09:20:19:506] AT+BLEGATTCPRIMSRV=0
```
And then my ESP32 rebotos. Console log is:
```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited)
. Exception was unhandled.
Register dump:
PC      : 0x4013be1f  PS      : 0x00060630  A0      : 0x8013fbda  A1      : 0x3ffe7c00  
A2      : 0x00000000  A3      : 0x3ffc7490  A4      : 0x3ffd59a8  A5      : 0x3ffd59a8  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x8013be5c  A9      : 0x3ffe7be0  
A10     : 0x3ffd597c  A11     : 0x40083e24  A12     : 0x3ffffe0c  A13     : 0x3ffe7c38  
A14     : 0x00000000  A15     : 0x3ffffe28  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000004  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x4013be1f:0x3ffe7c00 0x4013fbd7:0x3ffe7c20 0x4013b1ad:0x3ffe7c70 0x40142eb6:0x3ffe7c90

Rebooting...
```
The GATT profile of [SensorTAG CC650](http://www.ti.com/tool/cc2650stk) could be found here: 

- http://processors.wiki.ti.com/index.php/CC2650_SensorTag_User%27s_Guide#Gatt_Server

- http://processors.wiki.ti.com/images/a/a8/BLE_SensorTag_GATT_Server.pdf

So, which is the problem? Maybe it's because the GATT profile is not follows the standard defined here: https://www.bluetooth.com/specifications/gatt/generic-attributes-overview?

Thanks

xcguang, 2019-11-12:
Can you try the latest firmware?

carlessole, 2019-11-12:
Not yet.
I will try during this week.
Thanks!

carlessole, 2019-11-13:
Great! Now it's working ok.

```
[11:51:05:119] AT+BLEINIT=1␍␊
[11:51:05:699] ␍␊
[11:51:05:701] OK␍␊
[11:51:10:908] AT+BLECONN=0,"54:6c:0e:4d:53:03"␍␊
[11:51:11:059] +BLECONN:0,"54:6c:0e:4d:53:03"␍␊
[11:51:11:069] ␍␊
[11:51:11:069] OK␍␊
[11:51:12:954] AT+BLEGATTCPRIMSRV=0␍␊
[11:51:13:864] +BLEGATTCPRIMSRV:0,1,0x1800,1␍␊
[11:51:13:866] +BLEGATTCPRIMSRV:0,2,0x1801,1␍␊
[11:51:13:870] +BLEGATTCPRIMSRV:0,3,0x180A,1␍␊
[11:51:13:872] +BLEGATTCPRIMSRV:0,4,0x180F,1␍␊
[11:51:13:874] +BLEGATTCPRIMSRV:0,5,0xF000AA0004514000B000000000000000,1␍␊
[11:51:13:880] +BLEGATTCPRIMSRV:0,6,0xF000AA2004514000B000000000000000,1␍␊
[11:51:13:885] +BLEGATTCPRIMSRV:0,7,0xF000AA4004514000B000000000000000,1␍␊
[11:51:13:890] +BLEGATTCPRIMSRV:0,8,0xF000AA8004514000B000000000000000,1␍␊
[11:51:13:895] +BLEGATTCPRIMSRV:0,9,0xF000AA7004514000B000000000000000,1␍␊
[11:51:13:900] +BLEGATTCPRIMSRV:0,10,0xFFE0,1␍␊
[11:51:13:903] +BLEGATTCPRIMSRV:0,11,0xF000AA6404514000B000000000000000,1␍␊
[11:51:13:908] +BLEGATTCPRIMSRV:0,12,0xF000AC0004514000B000000000000000,1␍␊
[11:51:13:913] +BLEGATTCPRIMSRV:0,13,0xF000CCC004514000B000000000000000,1␍␊
[11:51:13:918] +BLEGATTCPRIMSRV:0,14,0xF000FFC004514000B000000000000000,1␍␊
[11:51:13:924] ␍␊
[11:51:13:924] OK␍␊
```

Thanks.
I close the issue!

closed 2019-11-13 by carlessole