espressif/esp-at #175: esp32 recive from mobile but cannot awnser

rogmadrid, 2019-02-11:
Hi, I'm trying to connect the device to my mobile pone using bluetooth and the app Serial Bluetooth Terminal, I have follow this steps:

Using the console of cutecom, connected to the device:
Set the device as server
at+bleinit=2
Create and start services
at+blegattssrvcre
at+blegattssrvstart
Set what data we send (needed specifically this one to connect with SBT)
At+bleadvdata="0201061aff4c000215fda50693a4e24fb1afcfc6eb0764782527b7f206c5"
Look what is the address of the device
At+bleaddr?
Config the serial port
At+blesppcfg=1,1,7,1,5 (this are the characteristics that we want to write and that we want to read from the client)
Start advising
At+bleadvstart

From this point I work in the app:
I look for my device press it and enter to edit, select custom, press at service uuid, the device will connect and give you his services to choose what you will listen and write. 

At that point I can send information to the device that is printed in the console, but I can't do the same in the other way, if i try, the device report me an error. How I can configure the app and the device to open a bidireccional comunication?

Thanks

xiewenxiang, 2019-02-12:
Hi ,
Before ESP32 sends data to the  phone, the phone needs to write the descrption of the 7th characteristic( Enable notification).
And then, Put ESP32 into SPP mode.
 * AT+BLESPP

rogmadrid, 2019-02-12:
Oh, thanks very much, i forgot to put the ESP32 into SPP mode.

Now everything is working nice, but now I can't close the communication without disconecting the terminal app or the device (I tried to send +++ as it's said in the manual, but nothing happen).

Thanks again for everything

closed 2019-11-12 by xcguang