espressif/esp-at #176: [Question] Is there a way to get network statistics?

sarai2, 2019-02-13:
I'm trying to receive a lot of UDP packet using AT command and seeing some packets drop.  Of course, it's UDP and I'm not expecting error-free communication, but I'm wondering if this is because 1) the packet was not received by the ESP32, or 2) ESP32 received the packet, but as I didn't read the data timely, it was lost.  If the latter is the case, I can do something with my software to reduce the number of lost packets.  Is there a way to find out if (or how much) data was lost due buffer overrun?

xcguang, 2019-02-20:
Hi,
Current AT command does not support this feature.

xcguang, 2019-11-12:
Hi,
I will close this issue, if you have any question, please reopen it again.

closed 2019-11-12 by xcguang