espressif/esp-at #178: ESP-IDF version used

carlessole, 2019-02-18:
Hello,
today a new version of ESP-IDF has been released: [v3.1.3](https://github.com/espressif/esp-idf/releases/tag/v3.1.3)
I know that right now the current version of this project, [v.1.1.3.0](https://github.com/espressif/esp32-at/releases/tag/v1.1.3.0), was not using the latest stable version of ESP-IDF but, is there any plan to finally bring the latest stable version of ESP-IDF into esp32-at project? And so make esp32-at compatible with ESP-IDF 3.1.x (instead of v3.0.x)?

Thank you

xcguang, 2019-02-20:
Hi，
We plan to skip this branch, and will release AT v1.2.0.0 based on ESP_IDF v3.2.

carlessole, 2019-02-20:
Thanks @xcguang .
And, is there an estimation of when this new version will be released?

xcguang, 2019-02-20:
Now test team is testing ESP-IDF v3.2, after ESP-IDF v3.2 is released, we will test AT v1.2.0.0 and release it. If everything goes well, maybe in early March or mid-March.

carlessole, 2019-02-20:
@xcguang . Thank you very much!

carlessole, 2019-04-03:
Do you know about the estimated date for this release?
Thanks

closed 2019-07-11 by carlessole