espressif/esp-at #182: STA IP Error

wnoliveira, 2019-02-25:
I was having troubles when trying to connect to a TCP sever. I don't know exactly how, but the source of my problems is that the STA IP is set to "0.0.0.0". I tried to rewrite this IP, to a valid one, but without success. If you notice, the command AT+CIFSR keeps returning the IP "0.0.0.0".

The only way to return to a valid IP was sending a RESTORE command, as shown in the terminal below.

[image]

I think that it's a bug on firmware. Does anyone have another idea?

SONGjiaxiu, 2019-07-31:
Hi,
The following is LOG.The image shows the version information for the esp32-at project I used.
[image]
Which branch of the esp32-at program do you use?
thanks

xcguang, 2019-11-12:
Close it because of no update for a long time, if you have any question, please reopen it again.

closed 2019-11-12 by xcguang