espressif/esp-at #183: Not booting after flash

glitch911, 2019-02-26:
Hi all,

I must be missing something but need some hints about what :)

I just cloned the repository esp32-at and then :
make menuconfig (change the serial port)
make all
got some warning :
`$ make -j12 all
Makefile:29: There is no module_config/module_wroom-32,Using module_config/module_esp32_default
fatal: destination path '/home/usr/esp/esp-idf' already exists and is not an empty directory.
HEAD is now at 646d36c72 Merge branch 'feature/nmea-parser' into 'master'

/home/usr/esp/esp32-at/tools/mkfatfs/Makefile.projbuild:13: warning: overriding recipe for target 'clean'
/home/usr/esp/esp-idf/components/app_update/Makefile.projbuild:41: warning: ignoring old recipe for target 'clean'
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0
GENCONFIG
App "esp-at" version: v1.0.0.0-203-gcecb29c
Makefile:29: There is no module_config/module_wroom-32,Using module_config/module_esp32_default
fatal: destination path '/home/usr/esp/esp-idf' already exists and is not an empty directory.
HEAD is now at 646d36c72 Merge branch 'feature/nmea-parser' into 'master'

/home/usr/esp/esp32-at/tools/mkfatfs/Makefile.projbuild:13: warning: overriding recipe for target 'clean'
/home/usr/esp/esp-idf/components/app_update/Makefile.projbuild:41: warning: ignoring old recipe for target 'clean'
building AT customized partition table module_config/module_esp32_default/at_customize.csv
`
make flash

-> flash is OK - disconnect BOOT0 from GND - and when reset got this on UART0 :
`                                          
                                    
Rebooting...
                                                                  
ets Jun  8 2016 00:22:57
                                                      
                          
                                                    
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
                        
configsip: 0, SPIWP:0xee
                                                      
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
       
mode:DIO, clock div:2
                                                         
load:0x3fff0018,len:4
                                                         
load:0x3fff001c,len:6196
                                                      
load:0x40078000,len:9968
                                                      
load:0x40080400,len:6532
                                                      
entry 0x40080760
                                                              
[0;32mI (29) boot: ESP-IDF v3.3-beta1 2nd stage bootloader[0m
               
[0;32mI (29) boot: compile time 11:33:50[0m
                                 
[0;32mI (29) boot: Enabling RNG early entropy source...[0m
                  
[0;32mI (34) boot: SPI Speed      : 40MHz[0m
                                
[0;32mI (38) boot: SPI Mode       : DIO[0m
                                  
[0;32mI (42) boot: SPI Flash Size : 4MB[0m
                                  
[0;32mI (46) boot: Partition Table:[0m
                                      
[0;32mI (50) boot: ## Label            Usage          Type ST Offset   Length[
0m
                                                                            
[0;32mI (57) boot:  0 phy_init         RF data          01 01 0000f000 00001000
[0m
                                                                          
[0;32mI (64) boot:  1 otadata          OTA data         01 00 00010000 00002000
[0m
                                                                          
[0;32mI (72) boot:  2 nvs              WiFi data        01 02 00012000 0000e000
[0m
                                                                          
[0;32mI (79) boot:  3 at_customize     unknown          40 00 00020000 000e0000
[0m
                                                                          
[0;32mI (87) boot:  4 ota_0            OTA app          00 10 00100000 00180000
[0m
                                                                          
[0;32mI (94) boot:  5 ota_1            OTA app          00 11 00280000 00180000
[0m
                                                                          
[0;32mI (102) boot: End of partition table[0m
                               
[0;32mI (106) boot: No factory image, trying OTA 0[0m
                       
[0;32mI (111) esp_image: segment 0: paddr=0x00100020 vaddr=0x3f400020 size=0x32
b28 (207656) map[0m
                                                          
[0;32mI (193) esp_image: segment 1: paddr=0x00132b50 vaddr=0x3ffbdb60 size=0x03
074 ( 12404) load[0m
                                                         
[0;32mI (198) esp_image: segment 2: paddr=0x00135bcc vaddr=0x3ffc0bd4 size=0x00
000 (     0) load[0m
                                                         
[0;32mI (199) esp_image: segment 3: paddr=0x00135bd4 vaddr=0x40080000 size=0x00
400 (  1024) load[0m
                                                         
[0;32mI (209) esp_image: segment 4: paddr=0x00135fdc vaddr=0x40080400 size=0x0a
034 ( 41012) load[0m
                                                         
[0;32mI (234) esp_image: segment 5: paddr=0x00140018 vaddr=0x400d0018 size=0xfe
c58 (1043544) map[0m
                                                         
[0;32mI (600) esp_image: segment 6: paddr=0x0023ec78 vaddr=0x4008a434 size=0x0b
b80 ( 48000) load[0m
                                                         
[0;32mI (620) esp_image: segment 7: paddr=0x0024a800 vaddr=0x400c0000 size=0x00
064 (   100) load[0m
                                                         
[0;32mI (621) esp_image: segment 8: paddr=0x0024a86c vaddr=0x50000000 size=0x00
000 (     0) load[0m
                                                         
[0;32mI (627) esp_image: segment 9: paddr=0x0024a874 vaddr=0x50000000 size=0x00
000 (     0) load[0m
                                                         
[0;32mI (649) boot: Loaded app from partition at offset 0x100000[0m
         
[0;32mI (649) boot: Disabling RNG early entropy source...[0m
                
C:/msys32/home/tlebivic/esp/esp-idf/components/freertos/queue.c:1441 (xQueueGene
ricReceive)- assert failed!
                                                   
abort() was called at PC 0x40094879 on core 0
                                 
                                               
                               
Backtrace: 0x4009069c:0x3ffbc5c0 0x400908a9:0x3ffbc5e0 0x40094879:0x3ffbc600 0x4
00d445e:0x3ffbc640 0x4009409d:0x3ffbc670
                                      
                                          
                                    
Rebooting...
  
`
Any idea ?

For information with latest SDK ESP32-WROOM-32 AT Bin V1.1.2 flashed with Flash download tool all is working great.

vagr02, 2019-03-05:
I have same problems with last build :(

zhoufs, 2019-03-07:
I have same issue too.

NemesisXB, 2019-03-15:
Same issue here.

wjj-chn, 2019-03-22:
Please checkout to master version ,and update the code form github.
command refer:
git pull
git submodule update --init

zhoufs, 2019-03-26:
It works now, thanks!

langestefan, 2019-03-28:
I still have this exact issue. It's completely blocking me from working with esp32-at. Any update/fix? @wjj-chn. I tried:

> git pull
> git submodule update --init

but no change...

NemesisXB, 2019-03-29:
@langestefan I checked out the V1.1.3.0 code and used that. For some reason checking out esp32-at and reverting to the specific commit also resulted in a rebooting image so downloaded the code manually:
(On Linux):

`cd ~`

```
wget https://github.com/espressif/esp32-at/archive/v1.1.3.0.tar.gz
tar -xzf v1.1.3.0.tar.gz
mv esp32-at-v1.1.3.0 esp-at
```
```
cd esp-at
git clone -b v3.0.7 --recursive https://github.com/espressif/esp-idf.git esp-idf
```

glitch911, 2019-03-29:
It works for me as well now. Thx for fix !

langestefan, 2019-03-29:
> @langestefan I checked out the V1.1.3.0 code and used that. For some reason checking out esp32-at and reverting to the specific commit also resulted in a rebooting image so downloaded the code [...]

Thanks, this solved all of the previous issues!

But unfortunately I now have another problem:

`320028010@YY151746 MINGW32 /home/esp-at
$ make flash
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0
CC build/bootloader/bootloader_support/src/bootloader_random.o
CC build/bootloader/bootloader_support/src/flash_encrypt.o
CC build/bootloader/bootloader_support/src/bootloader_sha.o
CC build/bootloader/bootloader_support/src/esp_image_format.o
CC build/bootloader/bootloader_support/src/flash_partitions.o
CC build/bootloader/bootloader_support/src/secure_boot_signatures.o
CC build/bootloader/bootloader_support/src/bootloader_clock.o
CC build/bootloader/bootloader_support/src/secure_boot.o
CC build/bootloader/bootloader_support/src/bootloader_common.o
CC build/bootloader/bootloader_support/src/bootloader_utility.o
CC build/bootloader/bootloader_support/src/bootloader_init.o
CC build/bootloader/bootloader_support/src/efuse.o
CC build/bootloader/bootloader_support/src/flash_qio_mode.o
CC build/bootloader/bootloader_support/src/bootloader_flash.o
AR build/bootloader/bootloader_support/libbootloader_support.a
CC build/bootloader/log/log.o
AR build/bootloader/log/liblog.a
CC build/bootloader/spi_flash/spi_flash_rom_patch.o
AR build/bootloader/spi_flash/libspi_flash.a
CC build/bootloader/micro-ecc/micro-ecc/uECC.o
AR build/bootloader/micro-ecc/libmicro-ecc.a
CC build/bootloader/soc/esp32/rtc_clk.o
CC build/bootloader/soc/esp32/rtc_time.o
CC build/bootloader/soc/esp32/rtc_sleep.o
CC build/bootloader/soc/esp32/gpio_periph.o
CC build/bootloader/soc/esp32/rtc_clk_init.o
CC build/bootloader/soc/esp32/spi_periph.o
CC build/bootloader/soc/esp32/rtc_init.o
CC build/bootloader/soc/esp32/sdio_slave_periph.o
CC build/bootloader/soc/esp32/rtc_wdt.o
CC build/bootloader/soc/esp32/sdmmc_periph.o
CC build/bootloader/soc/esp32/cpu_util.o
CC build/bootloader/soc/esp32/rtc_periph.o
CC build/bootloader/soc/esp32/rtc_pm.o
CC build/bootloader/soc/esp32/soc_memory_layout.o
CC build/bootloader/soc/src//memory_layout_utils.o
AR build/bootloader/soc/libsoc.a
CC build/bootloader/main/bootloader_start.o
AR build/bootloader/main/libmain.a
LD build/bootloader/bootloader.elf
esptool.py v2.6
building AT customized partition table at_customize.csv
building AT customized partition table succeed
generating AT customized partition bins to /home/esp-at/build/customized_partitions
generating ble_data.bin: python /home/esp-at/tools/BLEService.py -t /home/esp-at/build/customized_partitions/ble_data.bin /home/esp-at/components/customized_partitions/raw_data/ble_data/example.csv
Traceback (most recent call last):
  File "C:/msys32/home/esp-at/tools/BLEService.py", line 29, in <module>
    import yaml
ImportError: No module named yaml
make: *** [/home/esp-at/components/customized_partitions/Makefile.projbuild:43: /home/esp-at/build/at_customize.bin] Error 1
`

Any idea on how to fix this?

NemesisXB, 2019-03-29:
> Thanks, this solved all of the previous issues!
>
> But unfortunately I now have another problem:
>
> `320028010@YY151746 MINGW32 /home/esp-at $ make flash Toolchain path: [...]

You are missing the pyyaml module:
try:
`pip install pyyaml`

langestefan, 2019-03-31:
@NemesisXB 
> You are missing the python yaml module:
> try:
> `pip install pyyaml`

Thanks a bunch! Everything up and running now. Really curious though as to why this release is not functioning properly. Maybe someone from Espressif can elaborate?

closed 2019-11-12 by xcguang