espressif/esp-at #187: How to enable bt classic in esp32 wroom

alekarasma, 2019-03-07:
Hello Everyone,

I am using AT commands to connect with esp32. I am able to use ble and wiifi commands but when i use bt classic commands it is giving error.

I used example commands from https://github.com/espressif/esp32-at/b ... uetooth.md
for BT classic
its giving error.

Do I need to enable BT classic seperately. if yes how and from where?
Is there any documentation to how to enabled BT Classic?
and also AT commands doc for BT classic?

More info:
i am using the latest AT firmware version from `ESP- ESP32-WROOM-32 AT Bin V1.1.2

`Thanks,
Asma

vagr02, 2019-03-12:
As far as i know there in regular firmware BT Classic is disabled. You need to install development tools, clone  esp32-at project repository , enable BT classic support and commands in menuconfig, and then  compile and flash this new firmware. BT classic works then, at least in my case ;-)

xcguang, 2019-11-12:
BT Classic is enabled in ESP32-WROVER-32 AT Bin V2.0.0 at https://www.espressif.com/en/support/download/at

closed 2019-11-12 by xcguang