espressif/esp-at #189: UDP frames not received while BLESCAN

carlessole, 2019-03-13:
Hello,
In esp32-at version 1.1.3.0 (last release) when listening to a port in UDP everything works fine and every UDP message is caught by ESP32.
```
AT+CWMODE=1

OK
AT+CWJAP="MySSID","MyPassword"
WIFI CONNECTED
WIFI GOT IP

OK
AT+CIPMUX=1

OK
AT+CIPSTART=4,"UDP","255.255.255.255",3310,3310,0
4,CONNECT

OK

+IPD,4,10:MyMessage

```

But when I add a BLESCAN with following parameters. Most of the times, UDP message is lost (just 1/5 is received aprox).
```
AT+CWMODE=1

OK
AT+CWJAP="MySSID","MyPassword"
WIFI CONNECTED
WIFI GOT IP

OK
AT+CIPMUX=1

OK
AT+CIPSTART=4,"UDP","255.255.255.255",3310,3310,0
4,CONNECT

OK
AT+BLEINIT=1

OK
AT+BLESCANPARAM=0,0,0,30,23

OK
AT+BLESCAN=1

OK
+BLESCAN:17:64:fa:bf:a8:14,-79,1eff060001092002ae852b71400d1cfbb514e573775fcd0bbf0441aa295b26,,1
+BLESCAN:05:25:c1:bf:2c:db,-89,1eff0600010920025e7741b713de5946f645e28a10a9067a3c528d107d5b64,,1
+BLESCAN:17:64:fa:bf:a8:14,-77,1eff060001092002ae852b71400d1cfbb514e573775fcd0bbf0441aa295b26,,1
+BLESCAN:1a:a5:17:37:eb:9a,-72,1eff0600010920025b7db8470fd82d41ff4bcea51aa9904688b5132a3fa487,,1
+BLESCAN:0d:71:c7:ac:8d:53,-83,1eff060001092002ac34537569bd2cb90eb1c55236c05ab0b2645161d213b0,,1
+BLESCAN:05:25:c1:bf:2c:db,-85,1eff0600010920025e7741b713de5946f645e28a10a9067a3c528d107d5b64,,1
(....)

```
Is it a bug? Am I doing anything wrong?

Thanks

xiewenxiang, 2019-09-18:
Wifi performance will decrease when BLE is working.

xcguang, 2019-11-12:
Please try the latest firmware, which optimizes wifi and bluetooth coexistence

carlessole, 2019-11-13:
I have tried with the new version [v2.0.0.0_esp32](https://github.com/espressif/esp-at/releases/tag/v2.0.0.0_esp32).
Being my settings:

> [11:33:05:991] AT+CIPSTA?␍␊
> [11:33:05:993] +CIPSTA:ip:"192.168.1.100"␍␊
> [11:33:05:997] +CIPSTA:gateway:"192.168.1.254"␍␊
> [11:33:06:001] +CIPSTA:netmask:"255.255.255.0"␍␊
> [11:33:06:001] ␍␊
> [11:33:06:005] OK␍␊

It is still not working when doing broadcasting:
[image]

Is UDP is intented for unicast, it works:
[image]

Thanks.

xiewenxiang, 2020-01-09:
In the example you provide, BLE takes up most of the RF work time.
It is recommended to set the scan parameters to reduce the occupancy of BLE.

closed 2021-04-14 by FayeY