espressif/esp-at #192: AWS put fails

cwq9999, 2019-04-02:
Able to connect to AWS via SSL using these commands:

```
AT> AT+CIPSSLCCONF=0,0
AT? OK%n
AT< 
AT< AT+CIPSSLCCONF=0,0
AT= OK
AT> AT+CIPSTART=0,"SSL","bucket.s3-us-west-2.amazonaws.com",443
AT? OK%n
```
However, I can't seem to get the signatures to match when I send a string like this:

```
sprintf(sbufferFile,"PUT /%s HTTP/1.1\r\nContent-Length: %d\r\nContent-Type: text/plain\r\nHost: bucket.s3-us-west-2.amazonaws.com\r\nX-Amz-Content-Sha256: %s\r\nX-Amz-Date: %s\r\nAuthorization: AWS4-HMAC-SHA256 Credential=%s/%s/us-west-2/s3/aws4_request, SignedHeaders=content-length;content-type;host;x-amz-content-sha256;x-amz-date, Signature=%s\r\n\r\n%s", upload_file_name, payload_len, payloadSHA256, fulltime_str, api_access, date_stamp, signature_print, payload);
```

The signature matches output from other tools such as python, so was wondering if this is:

a) a string formatting issue

or 

b) if something to do with certificates

Was hoping someone might have some insight into how to achieve this via AT commands.

xiewenxiang, 2019-04-02:
@cwq9999 

https://esp32.com/viewtopic.php?f=2&t=9374

cwq9999, 2019-04-02:
@xiewenxiang  Thank you for providing this link. Do you know if there is a way to publish directly to a s3 bucket rather than needing to use the iot route? Perhaps it all ends up in the same place, but I'm actually trying to push to digitalocean (which is bases on AWS), but I don't believe they provide the iot option as AWS does.

ubichsel, 2019-05-08:
Probably it is a problem with port 443. AWS needs a special option during the TLS handshake called Application-Layer Protocol Negotiation Extension. With this extension Amazon can distinguish between different protocols on the same port like HTTP / MQTT ...

Here a link from Amazon:
[https://docs.aws.amazon.com/iot/latest/developerguide/protocols.html](https://docs.aws.amazon.com/iot/latest/developerguide/protocols.html)

With dedicated different ports it is working but port 443 is usually better for going through firewalls etc.

Unfortunately it is not supported over AT-Commands right now (as of my knowledge). I would appreciate if Espressif would implement this feature (e.g. as an optional parameter in AT+CIPSSLCCONF param).

xcguang, 2019-11-12:
We have a guidance  to connect AWS using `AT+MQTT`, but it is in Chinese. https://blog.csdn.net/espressif/article/details/101714951

xcguang, 2019-11-12:
I will close this issue, if you have any question, please reopen it again.

closed 2019-11-12 by xcguang