espressif/esp-at #193: include file "uECC.h" missing

fantasiiio, 2019-04-05:
Hi,

I'm following instructions on https://github.com/espressif/esp32-at
I'm on windows and installed toolchain using this guide: https://docs.espressif.com/projects/esp-idf/en/latest/get-started/windows-setup.html#introduction

- 'make defconfig' worked, with a wvroom32 library not found warning.
- 'make menuconfig' worked
- But 'make flash' failed. I attached a screenshot.
[image]

Is there something I can do ?

imxood, 2019-04-08:
in ubuntu, the same

imxood, 2019-04-08:
Hi, i test a method, it is OK.
1. git clone -b release/v1.2.0.0 https://github.com/espressif/esp32-at.git
    cd esp32-at
2. change the .gitmodules, add one line:
    branch = release/v3.3
    then, run command:
        git submodule update --init --recursive
3. make menuconfig
    make flash -j12

fantasiiio, 2019-04-08:
It seem to have compiled without error. Thanks, that's awesome !

But now... Does the firmware is uploaded to the mcu ? There was no progress bar of erasing memory and flashing... Is it normal ?

imxood, 2019-04-09:
It work well. After "make flash -j12", You can find the wifi by your phone.
"make flash -j12", the command can build the esp-idf and esp-at, flash program to your esp32 device.

fantasiiio, 2019-04-09:
Classic bluetooth commands doesn't seem to be available in the "make menuconfig" in this version.. I need these commands.

xcguang, 2019-11-12:
Classic bluetooth command configuration  has added in `make menuconfig`

xcguang, 2019-11-12:
The lastest `ESP32-WROVER AT Bin V2.0` firmware has been released, which enables classic bluetooth command by default, you can find it at https://www.espressif.com/zh-hans/support/download/at

closed 2020-01-09 by xcguang