espressif/esp-at #195: AT_Classic_Bluetooth: Null characters (0x00) in UART-BT passthrough mode

iduquegit, 2019-04-12:
Hello,

I have an ESP32 configured and already connected to a smartphone in UART-BT passthrough mode. I'm able to send binary data (including 0x00) from the ESP32 to a terminal app but when the ESP32 receives the data it cuts the messages to the first 0x00 and ignores the rest.

I guess it's something related to strings and its termination, but I'm not familiarised with the AT firmware and I don't know where to start. Any ideas where I should start or modify the code?

Thank you very much in advance :)

caiubi, 2019-05-14:
Hi,
I have the same problem while receiving data in non passthrough mode. Did you find something that fixes it?

thales-nascimento, 2019-05-14:
I have the same problem. Sending these to esp32
"AT+BTSPPSEND=0,11\r\npudim\0pudim"
produce this imcomplete output
+BTDATA:12,pudim

mcaddia77, 2019-05-21:
I have the same problem.
I'm using AT command for BT classic by UART.
If i send to ESP32 data different to 0x00 it's all ok, but if in my data there are some NULL, the ESP32 don't sent to my device this character and for ESP32 che data packet is finished.
In this way i lost data.
Someone has fixed this bug?

xiewenxiang, 2019-06-04:
Hi, @iduquegit @caiubi @thales-nascimento @mcaddia77 

The code merge will take some time. Please provide the version information `AT+GMR`. I will provided temporary lib to you.

thales-nascimento, 2019-06-04:
mine is:
AT+GMR
AT version:1.3.0.0-dev(776615a - Apr 23 2019 09:36:36)
SDK version:v3.3-beta1-1-ga3e169309
compile time:Jun  4 2019 08:23:18

iduquegit, 2019-06-05:
Hello
Thank you for the support @xiewenxiang, here is my version:
`AT+GMR
AT version:1.3.0.0-dev(b262039 - Apr 10 2019 09:53:36)
SDK version:v3.3-beta1
compile time:Apr 11 2019 13:25:25`

WebertSS, 2019-06-07:
Hello, 
I have the same issue on my project. Here is my version:

AT version:1.3.0.0-dev(872c8d8 - Jan 22 2019 06:24:15)
SDK version:v3.3-beta1
compile time:Mar 14 2019 16:46:59

Thanks.

xiewenxiang, 2019-06-14:
Hi,
Please use the latest master branch.
commit: 4976fc10358d45722d6788b900434b98105255ea

WebertSS, 2019-06-14:
Hi @xiewenxiang
Thank you very much for your help.

iduquegit, 2019-06-18:
Hi,
I'm not able to compile the latest version of the AT firmware in master branch with IDF "v3.3-beta1". I get the following error:

E:/msys32/home/iduque/esp/esp-idf/components/bootloader_support/src/secure_boot_signatures.c:22:18: fatal error: uECC.h: No such file or directory
compilation terminated.

JudeDalton, 2019-06-19:
Hi,
I had the same issue with sending 0x00 producing incomplete data as it was interpreted as a null terminator. Building the latest master branch changed the behaviour, but now 0x00 is skipped - so if I send "DATA\0DATA" I get DATADATA - is there a way to maintain the 0x00 within the full data?

caiubi, 2019-06-19:
Hi,
Thank you for your help @xiewenxiang. Your new version solved the problem for me.

caiubi, 2019-06-19:
@JudeDalton, 
Are you sure that the \0 is not being sent? Because it maybe just not printable for you.

JudeDalton, 2019-06-20:
Hi, my mistake - it is working now, thank you...

iduquegit, 2019-06-20:
Hello,
@xiewenxiang: Thank you very much! I've test it and it works perfect now, I'm closing this issue.
@JudeDalton: Have you tried a terminal with hexadecimal view (as YAT or Termite)? Sending `DATA\0DATA` to the ESP you should get `0x44 0x41 0x54 0x41 0x00 0x44 0x41 0x54 0x41`

closed 2019-06-20 by iduquegit