espressif/esp-at #200: Wifi handshake timeout too short? BUG? 1sec instead of 10sec?

3s1d, 2019-04-25:
Hi,

I investigated the fact that it is almost impossible to connect to my iPhone's hotspot. Every other access point just works fine.

```
AT+CWJAP="Carmens iPhone","somePassword"
+CWJAP:4

ERROR
```

I digged a little bit deeper and found that:
```
I (138838) wifi: state: assoc -> run (10)
I (138842) wifi: start 10s connect timer for 4 way handshake
I (138848) wifi: wpa_psk start
I (138852) wifi: wpa_psk handle succeed
I (148847) wifi: handshake timeout t=8d31a61
I (148847) wifi: state: run -> init (200)
```
Note: Timeout after 1sec instead of the advertized 10sec.

A successful connection looks like that (same iPhone):
```
I (106687) wifi: state: assoc -> run (10)
I (106691) wifi: start 10s connect timer for 4 way handshake
I (106697) wifi: wpa_psk start
I (106701) wifi: wpa_psk handle succeed
I (106712) wifi: wpa_psk start
I (106715) wifi: wpa_psk handle succeed
I (106718) wifi: connected with Carmens iPhone, channel 1, bssid = e6:61:8b:0d:dc:8d
```
Note: Handshaking done in under 1sec.

So to me it seems that the phone is handling the handshake a little bit slow in most cases.
Anyway. A 10sec timer is advertized in the log but it times out after 1sec? I guess a zero is missing in the code? Unfortunately I can not find the responsible code...

Cheers
Juergen

Full log for reference:
```
I (136393) wifi: Start wifi disconnect
D (136394) phy_init: wifi mac time delta: 83697[0m
I (136396) wifi: connect step 1 -> 0, status=5, err_time=0
I (136396) wifi: connect status 5 -> 0, step=0, err_time=0
I (136401) wifi: state: run -> init (0)
I (136405) wifi: stop beacon/connect timer, send diassoc(8)
I (136410) wifi: sta leave
I (136412) wifi: pm stop, total sleep time: 7136776 us / 9084518 us

I (136418) wifi: stop CSA timer
I (136422) wifi: remove b4:fb:e4:41:3c:a3 from rc list
I (136426) wifi: new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (136433) wifi: filter: set rx policy=8
I (136436) wifi: Send disconnect event, reason=8, AP number=0
I (136442) wifi: filter: set rx policy=8
D (136447) event: SYSTEM_EVENT_STA_DISCONNECTED, ssid:Uranos, ssid_len:6, bssid:b4:fb:e4:41:3c:a3, reason:8,wifi reason: assoc leave[0m
V (136457) event: enter default callback[0m
V (136462) tcpip_adapter: check: local, if=0 fn=0x401bb4c8
[0m
D (136467) tcpip_adapter: if0 start ip lost tmr: enter[0m
D (136472) tcpip_adapter: if0 start ip lost tmr: already started[0m
V (136478) tcpip_adapter: call api in lwip: ret=0x0, give sem[0m
V (136484) tcpip_adapter: check: remote, if=0 fn=0x401bb4c8
[0m
V (136490) event: exit default callback[0m
I (136495) wifi: clear blacklist
D (136497) nvs: nvs_set_blob sta.ssid 36[0m
D (136503) nvs: nvs_set_blob sta.pswd 65[0m
[... 294 lines cut ...]
I (151396) wifi: wpa_psk start
I (151399) wifi: wpa_psk handle succeed
I (153853) wifi: Start wifi disconnect
I (153853) wifi: connect step 1 -> 0, status=1, err_time=0
I (153853) wifi: connect status 1 -> 0, step=0, err_time=0
I (153856) wifi: state: run -> init (0)
I (153859) wifi: stop beacon/connect timer, send diassoc(8)
I (153865) wifi: sta leave
I (153867) wifi: stop CSA timer
I (153870) wifi: remove e6:61:8b:0d:dc:8d from rc list
I (153875) wifi: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (153881) wifi: filter: set rx policy=8
I (153885) wifi: Send disconnect event, reason=8, AP number=0
I (153890) wifi: filter: set rx policy=8
D (153894) event: SYSTEM_EVENT_STA_DISCONNECTED, ssid:Carmens iPhone, ssid_len:14, bssid:e6:61:8b:0d:dc:8d, reason:8,wifi reason: assoc leave[0m
V (153907) event: enter default callback[0m
V (153911) tcpip_adapter: check: local, if=0 fn=0x401bb4c8
[0m
D (153916) tcpip_adapter: if0 start ip lost tmr: enter[0m
D (153921) tcpip_adapter: if0 start ip lost tmr: already started[0m
V (153928) tcpip_adapter: call api in lwip: ret=0x0, give sem[0m
V (153933) tcpip_adapter: check: remote, if=0 fn=0x401bb4c8
[0m
V (153939) event: exit default callback[0m
```

3s1d, 2019-04-25:
I updated esp-idf and esp32-wifi-lib to the current master. This seems to help a little bit. But still in more than 90% of the cases step 3 and 4 of the 4way handshake seem to fail... 

The timeout may be correct. I consider the 4way handshake to be broken with my iPhone...

SONGjiaxiu, 2019-07-31:
Hi,
when you fail to connect to the your iPhone, you can try to connect to the your iPhone with another device (PC or iPhone), what will happen?
thanks

xcguang, 2019-09-05:
Close because of no reply for a long time.

closed 2019-09-05 by xcguang