espressif/esp-at #201: 2Mb

tcpipchip, 2019-05-02:
Sir, can you help me how to config the two partitions files (at_customize.csv) and (partition_at.csv) to work with U-BLOX NINA-W102.
NINA W102 has 2MB flash!
Thanks!

PS: the bin size of AT COMMAND today is 1.3MB

sigalabs, 2019-05-03:
i have the same issue with UBLOX NINA-W102... i am trying to find a solution

tcpipchip, 2019-05-03:
Wow, good! Kostas
I did all that is possible...after try 30 times with different configurations i aborted!
Let me see if you get success

Enviado do meu iPhone

tcpipchip, 2019-05-03:
Kostas, did you get progress ?

sigalabs, 2019-05-06:
no success till now... i am trying to get rid of the OTA partitions to make room for the at_firmware but i get messages about overlaping addresses when running make flash command

tcpipchip, 2019-05-06:
Hi, i am having the same messages, the overlapping address!
We have to try to remove the FS size

keithduncan, 2019-05-20:
I was able to get this working. Here’s a summary of what I did, let me know if this helps 😄 

I used the Arduino MKR WiFi 1010 and wanted to replace the default [nina-fw](http://github.com/arduino/nina-fw) with the esp32-at firmware.

1. Load the Arduino MKR WiFi 1010 board with the SerialNINAPassthrough example
2. `git clone --recursive git@github.com:espressif/esp32-at`
3. cd esp32-at
4. Ensure the bundled esp-idk is used
    1. `unset IDK_PATH`
5. Remove the tx,rx,cts,rts pin override in `main/interface/uart/at_uart_task.c` or fix up the factory_params
    1. See https://github.com/keithduncan/esp32-at/commit/178317a87e823f40a2970f5fc5fd7802703dc709 for an example
6. Ensure the python dependencies are installed for the bundled esp-idk
    1. sudo pip install -r esp-idk/requirements.txt
7. Edit module_config/module_esp32_default/partitions_at.csv
    1. Remove the ota_0 and ota_1 partitions
    2. Add a factory,app,factory,0x100000,0x100000 partition
    3. See https://github.com/keithduncan/esp32-at/commit/178317a87e823f40a2970f5fc5fd7802703dc709 for an example
8. make menuconfig
    1. Setup the Serial flasher config
        1. Set flash size to 2MB
    2. Setup the AT commands port
        1. UART port 0, RX pin 3, TX pin 1, RTS pin -1 CTS pin -1
        2. Disable UART flow control
        3. Configure the required AT commands support
    3. See https://github.com/keithduncan/esp32-at/commit/22d1573da25d4d6cc6275182d31ee404f6d5b193 for an example config
9. `make flash simple_monitor`
    1. The chip should flash over the passthrough sketch
    2. You should now be able to send AT commands to the chip from your terminal

I figured most of this out from trial and error. http://bbs.esp32.com/viewtopic.php?t=8821 was also really helpful, it pointed me towards disabling hardware flow control (just setting the pins to -1 isn’t enough).

Good luck!

sigalabs, 2019-05-20:
> I was able to get this working. Here’s a summary of what I did, let me know if this helps 😄
> 
> I used the Arduino MKR WiFi 1010 and wanted to replace the default [nina-fw](https://github.com/arduino/nina-fw) with the esp32-at firmware.
> 
> 1. Load the Arduino MKR WiFi 1010 board with the SerialNINAPassthrough example
> 2. `git clone --recursive git@github.com:espressif/esp32-at`
> 3. cd esp32-at
> 4. Ensure the bundled esp-idk is used
>    
>    1. `unset IDK_PATH`
> 5. Remove the tx,rx,cts,rts pin override in `main/interface/uart/at_uart_task.c` or fix up the factory_params
>    
>    1. See [keithduncan@178317a](https://github.com/keithduncan/esp32-at/commit/178317a87e823f40a2970f5fc5fd7802703dc709) for an example
> 6. Ensure the python dependencies are installed for the bundled esp-idk
>    
>    1. sudo pip install -r esp-idk/requirements.txt
> 7. Edit module_config/module_esp32_default/partitions_at.csv
>    
>    1. Remove the ota_0 and ota_1 partitions
>    2. Add a factory,app,factory,0x100000,0x100000 partition
>    3. See [keithduncan@178317a](https://github.com/keithduncan/esp32-at/commit/178317a87e823f40a2970f5fc5fd7802703dc709) for an example
> 8. make menuconfig
>    
>    1. Setup the Serial flasher config
>       
>       1. Set the serial port to the correct path on your host
>       2. Set flash size to 2MB
>    2. Setup the AT commands port
>       
>       1. UART port 0, RX pin 3, TX pin 1, RTS pin -1 CTS pin -1
>       2. Disable UART flow control
>       3. Configure the required AT commands support
>    3. See [keithduncan@22d1573](https://github.com/keithduncan/esp32-at/commit/22d1573da25d4d6cc6275182d31ee404f6d5b193) for an example config
> 9. `make flash simple_monitor`
>    
>    1. The chip should flash over the passthrough sketch
>    2. You should now be able to send AT commands to the chip from your terminal
> 
> I figured most of this out from trial and error. http://bbs.esp32.com/viewtopic.php?t=8821 was also really helpful, it pointed me towards disabling hardware flow control (just setting the pins to -1 isn’t enough).
> 
> Good luck!

Hello! Thanks for the response,
i have tested it and its working, but it has dropped the BLE AT Commands support to make it fit to the flash.

if i enable BLE i get the same message

A fatal error occurred: File /home/kostas/esp32-at/build/esp-at.bin (length 1528656) at offset 1048576 will not fit in 2097152 bytes of flash. Use --flash-size argument, or change flashing address.
/home/kostas/esp32-at/esp-idf/components/esptool_py/Makefile.projbuild:62: recipe for target 'flash' failed

any idea if there could be a fix for this?

keithduncan, 2019-05-20:
You could try swapping the compiler optimisation level to release:

```
CONFIG_OPTIMIZATION_LEVEL_DEBUG=
CONFIG_OPTIMIZATION_LEVEL_RELEASE=y
```

That might generate a smaller binary.

If that isn’t enough I’d try reducing the size of the partitions in `at_customize.csv` for partitions you don’t need, `fatfs` is 576K so I’d start there, and updating the size of `at_customize` in `partitions_at.csv` accordingly.

tcpipchip, 2019-05-20:
Hello Duncan, 

Thanks too to try to reach the solution to memory size problem!

I agree, but, look if really dropped in your side the BLE AT commands!

Otherwise, i will compile again using your instructions, ok ?

Thank you again! It´s not easy!

tcpipchip, 2019-05-20:
I already tried to reduce the Partition but something still wrong!
Can you reduce the partition to 320K to see if work to us ?

keithduncan, 2019-05-20:
> I agree, but, look if really dropped in your side the BLE AT commands!

That’s correct, I’m not compiling with BLE support as I don’t need it.

> I already tried to reduce the Partition but something still wrong!

Can you post your partitions_at and at_customize partitions map? What error do you get when building your project?

tcpipchip, 2019-05-20:
just changed
on at_customize.csv to 320K
fatfs,data,fat,0x70000,**320K**
...but strange is that 0x70000 offset
and
on partition_at.csv
i removed the OTA and added
Add a factory,app,factory,0x100000,1M

the error is size is bigger that flash how was reported...

Can you test there ? When you have time!

Maybe reduce the value 0xE0000 - 256K ?

tcpipchip, 2019-05-20:
i lost more 2 hours without success 

very hard!

keithduncan, 2019-05-21:
I’m not able to build with BLE commands enabled. I’ve tried building with https://github.com/espressif/esp-idf/commit/646d36c7287d8e447b5bd76ec11071b739299887 and https://github.com/espressif/esp-idf/commit/8c57aa0242ad6e4b209129ef9cd2420224308257. Both give me errors like these at link time:

```
/Users/keithduncan/Projects/esp32-at/components/at/lib/libesp32_at_core.a(at_bleCmd.o):(.literal.at_gatts_connection_event_handler+0x20): undefined reference to `esp_ble_gap_disconnect'
```

---

Using `partitions_at.csv` and `at_customize.csv`:

```
# Name,   Type, SubType, Offset,  Size
phy_init, data, phy,     0xf000,  0x1000
otadata,data,ota,  0x10000,        8K
nvs,data,nvs,   0x12000,        56K
at_customize,0x40,0,0x20000,0x11000
factory,app,    factory,   0x40000,0x180000

# Name,Type,SubType,Offset,Size
ble_data,0x40,1,0x21000,12K
server_cert,0x40,2,0x24000,4K
server_key,0x40,3,0x25000,4K
server_ca,0x40,4,0x26000,4K
client_cert,0x40,5,0x27000,4K
client_key,0x40,6,0x28000,4K
client_ca,0x40,7,0x29000,4K
factory_param,0x40,8,0x2a000,4K
wpa2_cert,0x40,9,0x2b000,4K
wpa2_key,0x40,10,0x2c000,4K
wpa2_ca,0x40,11,0x2d000,4K
fatfs,data,fat,0x2f000,16K
```

It _compiles_ and passes the flash check but since it doesn’t link I can’t check the final binary size.

The app size of `0x180000` is more than the `1528656` binary size referenced in https://github.com/espressif/esp32-at/issues/201#issuecomment-493956688 ☝️ but I don’t know how big your binary is without seeing your sdkconfig or exact error message.

I also don’t know if having smaller at_customize partitions will cause issues (for you or the rest of the esp32-at firmware), it’s likely that you won’t need all of them and could drop the fatfs or wpa2 partitions, though I don’t know what dependencies there are on them in the firmware.

If the binary size is close to fitting I’d also encourage you to experiment with the compiler optimisation settings which may generate a smaller binary:

```
CONFIG_OPTIMIZATION_LEVEL_DEBUG=
CONFIG_OPTIMIZATION_LEVEL_RELEASE=y
```

tcpipchip, 2019-05-21:
I have a old esp-at sdk that compiles to ble and wifi. But doest not have too much resources! I guess that is a 1.2 version.

Btw, the MKR 1010, can you send us the compiled bin files ?

tcpipchip, 2019-05-21:
Hi,
I changed the parttions to
# Name,   Type, SubType, Offset,  Size
phy_init, data, phy,     0xf000,  0x1000
otadata,data,ota,  0x10000,        8K
nvs,data,nvs,   0x12000,        56K
at_customize,0x40,0,0x20000,0x11000
factory,app,    factory,   0x40000,0x180000

# Name,Type,SubType,Offset,Size
ble_data,0x40,1,0x21000,12K
server_cert,0x40,2,0x24000,4K
server_key,0x40,3,0x25000,4K
server_ca,0x40,4,0x26000,4K
client_cert,0x40,5,0x27000,4K
client_key,0x40,6,0x28000,4K
client_ca,0x40,7,0x29000,4K
factory_param,0x40,8,0x2a000,4K
wpa2_cert,0x40,9,0x2b000,4K
wpa2_key,0x40,10,0x2c000,4K
wpa2_ca,0x40,11,0x2d000,4K
fatfs,data,fat,0x2f000,16K

Compiled and got

-rw-r--r-- 1 osboxes osboxes    3072 May 21 12:16 at_customize.bin
-rw-r--r-- 1 osboxes osboxes 1273376 May 21 12:19 esp-at.bin
-rw-r--r-- 1 osboxes osboxes    8192 May 21 12:20 ota_data_initial.bin
-rw-r--r-- 1 osboxes osboxes    3072 May 21 12:16 partitions_at.bin
-rw-r--r-- 1 osboxes osboxes     144 May 21 12:19 phy_init_data.bin

I burned and
stop in the message
.
.
.
**boot: Disabiling RNG early entropy source...**

[image]

Something is wrong! But, we will find the solution!

Btw, i compiled that MKR 1010 
https://github.com/arduino/nina-fw
And i can burn on NINA W102.
Looks that is not AT command,  but a byte protocol level.
My question is, after burn this bin on NINA W102, can i use the Arduino Uno V2 to test it ?

tcpipchip, 2019-05-21:
Got this on nina-fw
[image]
is that right ?

tcpipchip, 2019-05-21:
looks be a SPI (HCI) communciation with this firwamre

keithduncan, 2019-05-22:
Yes the nina-fw firmware reads commands from an SPI port, that’s the firmware I replaced with esp32-at.

> Something is wrong! But, we will find the solution!

From the looks of your terminal output the firmware has booted but you aren’t getting the AT command prompt. How have you configured the `CONFIG_AT_UART_*` settings in the `esp32-at` `sdkconfig`? Make sure they are configured for the serial port you are connecting on.

tcpipchip, 2019-05-22:
I had used the default port! Did they change on version 1.3 ?
I will test late and report to us

tcpipchip, 2019-05-22:
Hi
did you you 
UART port 0, RX pin 3, TX pin 1, RTS pin -1 CTS pin -1
To NINA W102 ?
I see that TXD is 23 and RXD is 22

tcpipchip, 2019-05-23:
Thats all working now team :)

Thanks for all help Mr Duncan

[image]

keithduncan, 2019-05-23:
> UART port 0, RX pin 3, TX pin 1, RTS pin -1 CTS pin -1
> To NINA W102 ?
> I see that TXD is 23 and RXD is 22

Yep that’s the config I used 😄 

> Thats all working now team :)

That’s great news :tada: I’m glad you got it working :bow:

tcpipchip, 2019-05-23:
Yes! Working nice!

You are our hero! And you will help a lot of U-BLOX users :)

Thanks again!

tcpipchip, 2019-05-23:
sigalabs,

Did you see that works ?

sigalabs, 2019-05-24:
I can confirm that it works, i have tested both WIFI and BLE using AT commands.
Thank you very much @keithduncan and @tcpipchip for your time and support on this.
Nina W102 module, although it has 2MB flash, is still supported with these partition modifications.

tcpipchip, 2019-05-24:
Congratulations!

But @keithduncan  was the main mentor!

closed 2019-09-05 by xcguang