espressif/esp-at #202: esp32-at with esp-idf v3.2

carlessole, 2019-05-02:
Hello,
Almost 1 month ago, a new version of esp-idf was released: [v3.2](https://github.com/espressif/esp-idf/releases/tag/v3.2)
So is there a release plan for the esp32-at?
If yes, when it will be released?

Thanks!

xcguang, 2019-05-02:
Hi,
Sorry for that. We found a bug which blocks esp32-at release. We cannot release it before the new IDF which fixes this bug is released.

carlessole, 2019-05-02:
Thanks for the quick response.
And you can give an approximate date about when is estimated to be released (1month ahead, 3 months ahead...)
Thanks!

xcguang, 2019-05-02:
Sorry, it depends on the time IDF v3.2.1 releases. Maybe about a month.

carlessole, 2019-07-09:
Hello again,
Now that the release has been launched: https://github.com/espressif/esp32-at/tree/v1.2.0.0
It can be seen that the version of esp-idf is still v3.2.0, is that ok?

> - Update version number to ESP-IDF v3.2.0

Furthermore, in the [esp32-at](https://github.com/espressif/esp32-at/tree/v1.2.0.0) repo, the commit for the esp-idf submodule is not a commit from a release, it's just a commit in the branch (commit [ddbce78](https://github.com/espressif/esp-idf/tree/ddbce78bfff86dd65f9a7ee46b6e516461657dbf)):
[image]

So, this commit is the one that may be used for esp32-at project? Or it could be better to use an stable one (tag 3.2, 3.2.1 or 3.2.2).

Thank you

carlessole, 2019-07-26:
Hello,
anyone knows the answer?
Thank you!

xcguang, 2019-09-05:
@carlessole 
Hi, we have test all the AT function, it works well, because esp-idf always delays, we cannot wait it every time. So it may not be the release tag.

carlessole, 2019-09-05:
thanks @xcguang 
So I understand that the commit that must be used is that one = [ddbce78.](https://github.com/espressif/esp-idf/tree/ddbce78bfff86dd65f9a7ee46b6e516461657dbf)
Thanks!

closed 2019-09-24 by xcguang