espressif/esp-at #203: IOError with make flash

mattzilvetti, 2019-05-04:
I get this same error every time a try to run "make flash". I was able to flash the "Hello World" program just fine, but this keeps having issues. There seems to be a problem with the slashes between folders in the directory it's trying to reach. There was a similar issue (#157 ) but was closed before a solution was given. The terminal output is shown below:

Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0
Python requirements from C:/msys32/home/mattz/esp/esp-idf/requirements.txt are satisfied.

App "esp-at" version: v1.0.0.0-226-g13515da
Generating esp32.project.ld
Traceback (most recent call last):
  File "C:/msys32/home/mattz/esp/esp-idf/tools/ldgen/ldgen.py", line 122, in <module>
    main()
  File "C:/msys32/home/mattz/esp/esp-idf/tools/ldgen/ldgen.py", line 94, in main
    sdkconfig = SDKConfig(kconfig_file, config_file, args.env)
  File "C:/msys32/home/mattz/esp/esp-idf/tools/ldgen/sdkconfig.py", line 56, in __init__
    self.config = kconfiglib.Kconfig(kconfig_file)
  File "C:/msys32/home/mattz/esp/esp-idf/tools/kconfig_new/kconfiglib.py", line 656, in __init__
    self.top_node)  # prev_node
  File "C:/msys32/home/mattz/esp/esp-idf/tools/kconfig_new/kconfiglib.py", line 1583, in _parse_block
    self._enter_file(sourced_file)
  File "C:/msys32/home/mattz/esp/esp-idf/tools/kconfig_new/kconfiglib.py", line 1167, in _enter_file
    .format(self._filename, self._linenr, e.message))
IOError: C:/msys32/home/mattz/esp/esp-idf/Kconfig:59: Could not open 'C:msys32homemattzespesp-idfcomponentsapp_updateKconfig.projbuild' (ENOENT: No such file or directory). Perhaps the $srctree environment variable (which was unset) is set incorrectly. Note that the current value of $srctree is saved when the Kconfig instance is created (for consistency and to cleanly separate instances). Also note that e.g. $FOO in a 'source' statement does not refer to the environment variable FOO, but rather to the Kconfig Symbol FOO (which would commonly have 'option env="FOO"' in its definition).
make: *** [/home/mattz/esp/esp-idf/components/esp32/Makefile.projbuild:51: /home/mattz/esp/esp32-at/build/esp32/esp32.project.ld] Error 1

Neslon07, 2019-09-06:
Same as #230

xcguang, 2020-01-10:
Please try it using the latest master branch, if the issue persists, please reopen it again.

closed 2020-01-10 by xcguang