espressif/esp-at #205: AT+CIPSEND send to wrong IP address with UDP connection

ilcvm, 2019-05-09:
Test condition:
esp32 work in single connection mode (+CIPMUX=0) and UDP connection is created (AT+CIPSTART="UDP","192.168.101.110",1000,1002,2).

Issue condition:
Use command to start data transfer to remote IP_1:
**AT+CIPSEND=<length>,<IP_1>,<port>**
While sending data after "OK” and ">" is got, if a UDP packet is got
**+IPD,<len>,<IP_2>,<port>]:<data>**
Then after all data is sended and **Recv <length> bytes** **SEND OK**, the IP_2 will get the packet rather than IP_1

Issue reproduce confirmed AT firmware:

https://www.espressif.com/sites/default/files/ap/ESP32_AT_V1.1.3.zip
AT version:1.1.3.0(5a40576 - Nov 28 2018 12:50:55)
SDK version:v3.0.7
compile time:Dec 21 2018 09:04:56
Bin version:1.1.3(WROOM-32)

https://www.espressif.com/sites/default/files/ap/esp-wroom32_at_v1.0.zip
AT version:1.0.0.0(870285c - Nov 14 2017 07:26:38)
SDK version:v2.1-65-g8bca703
compile time:Nov 17 2017 07:53:25

https://github.com/espressif/esp32-at/tree/release/v1.2.0.0
AT version:1.2.0.0-dev(b4e03a6 - Apr 15 2019 07:24:21)

SONGjiaxiu, 2019-07-31:
Hi,
Do you mean "+IPD" is the reply of "AT+CIPSEND"?  
Thanks for your response.

fcl-a, 2019-08-02:
I need to communicate with different IP at the same time, so that "set the last parameter of the command to 0" can achive my purpose.

SONGjiaxiu, 2019-08-02:
Hi,
You can use "Multiple connections (AT+CIPMUX=1)" mode.
like:
First establish UDP transmission using the following command:
       AT+CIPMUX=1
       AT+CIPSTART=1,"UDP","192.168.1.100",8080,8001,0
       AT+CIPSTART=2,"UDP","192.168.1.105",8080,8002,0
Next, send the data using the following command:
       AT+CIPSEND=1,16,"192.168.1.100",8080
       AT+CIPSEND=2,6,"192.168.1.105",8080
Detailed content is available at https://github.com/espressif/esp32-at/releases.
Good luck.

fcl-a, 2019-08-02:
We require communicate with 8 IPs, but the max connection in current AT firmware can be created is only 5 (_<link ID>: ID of network connection (0~4), used for multiple connections_).

SONGjiaxiu, 2019-08-02:
Hi,
Since ESP32 or ESP8266 is limited by resources such as memory, the AT socket maximum  should be small. However, the AT project provides users with a configuration for the maximum number of socket connections. You can configure this parameter to be 8 or 10.As shown in the picture below.
[image]
BTW，You should also modify the "range" configuration for AT_SOCKET_MAX_CONN_NUM in the Kconfig file,and recompile and burn.
like:"range 1 10"
Try it,Good luck!

fcl-a, 2019-08-02:
@SONGjiaxiu 
Thank you very much. I will try that.

closed 2019-09-05 by xcguang