espressif/esp-at #206: Used UART0 for AT command but doesn't respond

Masi2017, 2019-05-24:
Hi all,
I am trying to use UART0 instead of UART1. I have changed the AT port in menuconfig (TXD0=1, RXD0=3 , CTS=22 , RTS=19). Also I've redirected debug port to UART1 and changed the UART pins to 16 and 17 for RX and TX. I have turned off flow control. I should add that I am using ESP32_DEV_KIT_V4 ( which has WROOM) . Unfortunately I am not able to communicate with that and I don't get any response from AT port. Also I've changed the factory_param_data too but nothing changed then I commented out those lines that set rx and tx based on partition file and hard forced it to use gpio 1 and 3 but nothingchanged .
 I've attached an screenshot here:
[image]

deadprogram, 2019-06-17:
Seems like this commit removed the ability to customize which UART is used for AT commands: https://github.com/espressif/esp32-at/commit/a8126c3e9c836b8e81f1c79de498f9424cf55cb5

This is actually a feature I really need!

xcguang, 2019-09-24:
Please refer to https://github.com/espressif/esp-at/blob/master/docs/How_To_Set_AT_Port_Pin.md, which we recently added.

closed 2019-11-12 by xcguang