espressif/esp-at #209: AT+BLESECPARAM

mrbiondo, 2019-06-07:
What are the settings that is possible to use with that commands?
Thanks

mrbiondo, 2019-06-07:
And, what does the parameters, <n> and <m>, mean in the message "+BLEAUTHCMPL:<n>,<m>" ?

alexdecei, 2019-06-13:
**The at commands documentation:**
https://www.espressif.com/sites/default/files/documentation/esp32_at_instruction_set_and_examples_en.pdf

**The meaning of the parameters:**
https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/gatt_security_server/tutorial/Gatt_Security_Server_Example_Walkthrough.md

**"+BLEAUTHCMPL"** means that the pairing is completed (it's in the at command documentation)

mrbiondo, 2019-06-13:
could you give me an example of "Secure Connections with MITM Protection" pairing, by using at commands?

alexdecei, 2019-06-13:
Sure:

`AT+BLESECPARAM=6,0,16,0,0` 
ESP_LE_AUTH_REQ_SC_MITM
ESP_IO_CAP_OUT
key size: 16
ESP_BLE_ENC_KEY_MASK
ESP_BLE_ENC_KEY_MASK

mrbiondo, 2019-06-13:
AT+GMR command says:
AT version:1.2.0.0-dev(6e68431 - Sep 21 2018 09:16:33)
SDK version:v3.1-34-gcea4d3132

AT+BLESECPARAM=6,0,16,0,0 says:
ERROR
Only by using 0 ,1, 4 as <auth_req> I get OK.

alexdecei, 2019-07-09:
Hi, same for me here: only 0,1,4 work for <auth_req>.
Working on it, I keep you in touch.

alexdecei, 2019-07-11:
Found nothing except that 8,9,12,13 works. That's weird.

AT+BLESECPARAM=8,1,16,3,3
AT+BLESECPARAM=8,1,16,3,3

alexdecei, 2019-07-11:
Found nothing except that 8,9,12,13 works. That's weird.

AT+BLESECPARAM=8,1,16,3,3
AT+BLESECPARAM=9,1,16,3,3

alexdecei, 2019-07-11:
Found nothing except that 8,9,12,13 works. That's weird.

`AT+BLESECPARAM=8,1,16,3,3`
_ok_
`AT+BLESECPARAM=9,1,16,3,3`
_ok_
`AT+BLESECPARAM=12,1,16,3,3`
_ok_
`AT+BLESECPARAM=13,1,16,3,3`
_ok_

mrbiondo, 2019-07-11:
In the commands documentation it is written that <auth_req>  ranges from 0 to 7.
So, what do 8,9,12,13 stand for?

alexdecei, 2019-07-11:
Yes that's concerning. I didn't test in detail those parameters but they work properly.

Rvice, 2020-01-07:
> Found nothing except that 8,9,12,13 works. That's weird.
> 
> `AT+BLESECPARAM=8,1,16,3,3`
> _ok_
> `AT+BLESECPARAM=9,1,16,3,3`
> _ok_
> `AT+BLESECPARAM=12,1,16,3,3`
> _ok_
> `AT+BLESECPARAM=13,1,16,3,3`
> _ok_

Change the second parameter to 0. If using 1, try setting the RANDOM ADDRESS before making this call.

xiewenxiang, 2020-03-19:
If you need MITM, You should to set the encryption parameters

`AT+BLESECPARAM=<auth_req>,<iocap>,<key_size>,<init_key>,<rsp_key>[,<auth_option>]`

the `auth_req` should set 4, 12 or 13.

Can't set  the `iocap` to 3.

closed 2020-03-19 by xiewenxiang