espressif/esp-at #212: How to build at_http_webserver

kzoltaan, 2019-06-25:
If I understand the sample, it makes esp32 with AT command support and with webpages . This is what we need!

The build of esp32-at with flashing is going very well, but I unable to compile at_http_webserver.
Get an error message:

make: *** No rule to make target '/esp32/esp32-at/at_customize.csv', needed by '/esp32/esp32-at/example/at_http_webserver/build/at_customize.bin'.  Stop.

Can somebody help me?

kzoltaan, 2019-07-17:
Here is my next problem. I made an update from repo, and get this:
`/esp32/esp32-at/example/at_http_webserver/build/atparse/libatparse.a(parseframe.o):(.literal.init_at_module+0x10): undefined reference to `esp_at_device_ops_regist'
/esp32/esp32-at/example/at_http_webserver/build/atparse/libatparse.a(parseframe.o):(.literal.init_at_module+0x14): undefined reference to `esp_at_module_init'
/esp32/esp32-at/example/at_http_webserver/build/atparse/libatparse.a(parseframe.o):(.literal.init_at_module+0x18): undefined reference to `esp_at_base_cmd_regist'
/esp32/esp32-at/example/at_http_webserver/build/atparse/libatparse.a(parseframe.o):(.literal.init_at_module+0x1c): undefined reference to `esp_at_wifi_cmd_regist'
/esp32/esp32-at/example/at_http_webserver/build/atparse/libatparse.a(parseframe.o):(.literal.init_at_module+0x20): undefined reference to `esp_at_net_cmd_regist'
/esp32/esp32-at/example/at_http_webserver/build/atparse/libatparse.a(parseframe.o):(.literal.init_at_module+0x24): undefined reference to `esp_at_ble_cmd_regist'
/esp32/esp32-at/example/at_http_webserver/build/atparse/libatparse.a(parseframe.o):(.literal.init_at_module+0x28): undefined reference to `esp_at_fs_cmd_regist'
/esp32/esp32-at/example/at_http_webserver/build/atparse/libatparse.a(parseframe.o):(.literal.at_call_lib+0x4): undefined reference to `esp_at_port_recv_data_notify'
/esp32/esp32-at/example/at_http_webserver/build/atparse/libatparse.a(parseframe.o): In function `init_at_module':
/esp32/esp32-at/example/at_http_webserver/components/atparse/parseframe.c:185: undefined reference to `esp_at_device_ops_regist'
/esp32/esp32-at/example/at_http_webserver/components/atparse/parseframe.c:186: undefined reference to `esp_at_module_init'
/esp32/esp32-at/example/at_http_webserver/components/atparse/parseframe.c:191: undefined reference to `esp_at_base_cmd_regist'
/esp32/esp32-at/example/at_http_webserver/components/atparse/parseframe.c:199: undefined reference to `esp_at_wifi_cmd_regist'
/esp32/esp32-at/example/at_http_webserver/components/atparse/parseframe.c:207: undefined reference to `esp_at_net_cmd_regist'
/esp32/esp32-at/example/at_http_webserver/components/atparse/parseframe.c:215: undefined reference to `esp_at_ble_cmd_regist'
/esp32/esp32-at/example/at_http_webserver/components/atparse/parseframe.c:223: undefined reference to `esp_at_fs_cmd_regist'
/esp32/esp32-at/example/at_http_webserver/build/atparse/libatparse.a(parseframe.o): In function `at_call_lib':
/esp32/esp32-at/example/at_http_webserver/components/atparse/parseframe.c:65: undefined reference to `esp_at_port_recv_data_notify'
collect2: error: ld returned 1 exit status
/esp32/esp32-at/example/at_http_webserver/../../esp-idf/make/project.mk:457: recipe for target '/esp32/esp32-at/example/at_http_webserver/build/esp-http-at.elf' failed
make: *** [/esp32/esp32-at/example/at_http_webserver/build/esp-http-at.elf] Error 1
`

kzoltaan, 2019-07-17:
other:
`usage: factory_param_generate.py [-h] [--platform PLATFORM] [--module MODULE]
                                 [--define_file DEFINE_FILE]
                                 [--module_file MODULE_FILE]
                                 [--bin_name BIN_NAME] [--log_file LOG_FILE]
factory_param_generate.py: error: argument --platform: expected one argument
/esp32/esp32-at/example/at_http_webserver/../../components/customized_partitions/Makefile.projbuild:38: recipe for target '/esp32/esp32-at/example/at_http_webserver/build/at_customize.bin' failed
make: *** [/esp32/esp32-at/example/at_http_webserver/build/at_customize.bin] Error 1`

kzoltaan, 2019-07-17:
just run:
`root@debian:/esp32/esp32-at/example/at_http_webserver# make`

SONGjiaxiu, 2019-07-18:
We have fixed this problem, see: //github.com/espressif/esp32-at/commit/d0f66f8ed8978fc5e126566880db704044558a8a，
You need to update the esp32-at project to the latest version, try it out.
thanks

SONGjiaxiu, 2019-07-18:
We have fixed this problem, see: https://github.com/espressif/esp32-at/commit/d0f66f8ed8978fc5e126566880db704044558a8a，
You need to update the esp32-at project to the latest version, try it out.
thanks

SONGjiaxiu, 2019-07-18:
We have fixed this problem, see: https://github.com/espressif/esp32-at/commit/d0f66f8ed8978fc5e126566880db704044558a8a
You need to update the esp32-at project to the latest version, try it out.
thanks

kzoltaan, 2019-07-18:
Working. Thanks for your help

closed 2019-09-05 by xcguang