espressif/esp-at #213: "fatal error: btc/btc_manage.h: No such file or directory" in esp32-at v1.2.0.0

carlessole, 2019-06-28:
Hello,
I was upgrading my application build on top of the esp32-at code with new version released v1.2.0.0.
I had this include before:

`#include "btc_manage.h"`

And when trying to compile the code when all changes imported I get this error

`fatal error: btc_manage.h: No such file or directory`

After looking the code on esp-idf used for this esp32-at release (commit [ddbce78bfff86dd65f9a7ee46b6e516461657dbf)](https://github.com/espressif/esp-idf/tree/ddbce78bfff86dd65f9a7ee46b6e516461657dbf) I see that "btc_manage.h" file has been moved from `components/bt/bluedroid->btc->include` to `components/bt/bluedroid->btc->include->btc`

So I tried to include the file using:

`#include "btc/btc_manage.h"`

But when compiling the same error occurs:

`fatal error: btc/btc_manage.h: No such file or directory`

Thank you.

Carles

xiewenxiang, 2019-06-28:
Probably the version of IDF you are using is error.

carlessole, 2019-06-28:
I don't think so...
I was using commit [ddbce78bfff86dd65f9a7ee46b6e516461657dbf](https://github.com/espressif/esp-idf/tree/ddbce78bfff86dd65f9a7ee46b6e516461657dbf) because it's the one used in the esp32-at v1.2.0.0 release:

[image]

Which version I have to use?
I also tried with the last esp-idf stable version: https://github.com/espressif/esp-idf/commits/v3.2.2, and the same issue is happening.

Thanks

carlessole, 2019-06-28:
I've found what caused the problem.
On `IDF_PATH/components/bt/component.mk` this new esp-idf has the following changes (compared to v3.0.7 that I used before)
[image]

Before all the path were publics and now they are under the "COMPONENT_PRIV_INCLUDEDIRS" variable.

I solved the issue adding this line (but I really don't like this solution):

`COMPONENT_ADD_INCLUDEDIRS += $(COMPONENT_PRIV_INCLUDEDIRS)`

Do you know if there is any other solution?
I need to call the `void *btc_profile_cb_get(btc_pid_t profile_id);` from my code to add my custom callback without overriding the ones set by lib_at.

```
	if(old_ble_gap_callback == NULL)
	{
		//First we get the current callback
		old_ble_gap_callback = (esp_gap_ble_cb_t)btc_profile_cb_get(BTC_PID_GAP_BLE);
		if(old_ble_gap_callback == NULL)
		{
			ESP_LOGD(_TAG, "%s: No existing BLE GAP callback...", __FUNCTION__);
		}
	}
	
	esp_err_t res = esp_ble_gap_register_callback(intesis_ble_gap_callback);
```

And the from my custom BLE callback I call it:

```
        if(old_ble_gap_callback != NULL)
	{
		 old_ble_gap_callback(event, param);
	}
```

Do you know if there is another way to achieve that using just the "public" includes?

`COMPONENT_ADD_INCLUDEDIRS +=    bluedroid/api/include/api`

Thanks

closed 2019-07-04 by xiewenxiang