espressif/esp-at #214: ESP32-PICO-D4 AT-commands on UART0 - How?

galyamovm, 2019-07-03:
Hello. I have an ESP32-PICO-D4 board and I want to use AT commands to control. From the documentation I concluded that only UART0 is available on this board, to which debugging information is sent by default and AT commands are not available. (UART1 and UART2 ports used for flash memory).
I downloaded the esp32-at.bin file to the chip at the appropriate address, but the AT commands did not appear (I also tried to load the factory_PICO-D4.bin file and this did not help either). I suspect that it is necessary to change the firmware so that the AT commands are available on the UART0 port, but I don’t understand how to do it yet.

How to control the chip using AT commands? Maybe there is a detailed instruction on how to enable AT commands on the UART0 port?

xiewenxiang, 2019-07-04:
You can use Uart0 to input the command, but Uart0 will have some unmasked logs.
If you really want to do this, you can modify 
`esp-at/components/customized_partitions/raw_data/factory_param/factory_param_data.csv`

Then recompile the project

```c
make
make print_flash_cmd | tail -n 1 > build/download.config
make factory_bin
```

galyamovm, 2019-07-04:
Thanks for the answer. Debug output can be turned off, or am I wrong?
And did I understand correctly that on the ESP32-PICO-D4 board, the control of AT commands is available only through the UART0 port?

xiewenxiang, 2019-07-04:
Debug information can be turned off, but some boot log cannot be completely turn off.

galyamovm, 2019-07-04:
Thanks, I will try.

galyamovm, 2019-07-11:
Hello, xiewenxiang!
I edited the parameters file and generated the firmware for the board, downloaded it - and so far nothing worked :(
[image]

In the Putty console, I see that the board is constantly rebooting:
[image]

I tried to change the settings by calling the command "make menuconfig" - screenshots attached.
[image]

[image]

Probably, I'm doing something wrong again?

swouf, 2019-07-31:
Hello !
I refresh the the issue as I am also doing this for an ESP32-WROOM. I did the same kind of change @galyamovm did but I couldn't manage to send AT commands to the ESP on its UART0 port. See issue #222 for more details about how I proceeded.

xcguang, 2019-09-24:
Maybe https://github.com/espressif/esp-at/blob/master/docs/How_To_Set_AT_Port_Pin.md is help to you.

closed 2019-11-12 by xcguang