espressif/esp-at #215: Build break while building esp32-at using release/v1.2.0.0 branch

bkvenkatesh, 2019-07-03:
[esp32-at-buildBreak.log](https://github.com/espressif/esp32-at/files/3357008/esp32-at-buildBreak.log)
I am getting a build break when using "make all" while compiling file asio.c on esp32-at on release/v1.2.0.0 branch. 

"CXX build/asio/asio/asio/src/asio.o
In file included from c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.2.0\cstdlib:75,
                 from c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.2.0\ext\string_conversions.h:41,
                 from c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.2.0\bits\basic_string.h:6391,
                 from c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.2.0\string:52,
                 from C:/msys32/home/workspace/esp32-at/esp-idf/components/asio/asio/asio/include/asio/impl/error.ipp:19,
                 from C:/msys32/home/workspace/esp32-at/esp-idf/components/asio/asio/asio/include/asio/impl/src.hpp:22,
                 from C:/msys32/home/workspace/esp32-at/esp-idf/components/asio/asio/asio/src/asio.cpp:11:
c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\sys-include\stdlib.h:155:44: error: expected initializer before '__result_use_check'
 void *reallocarray(void *, size_t, size_t) __result_use_check __alloc_size(2)
                                            ^~~~~~~~~~~~~~~~~~
c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\sys-include\stdlib.h:340:52: error: expected initializer before '__alloc_align'
 void * aligned_alloc(size_t, size_t) __malloc_like __alloc_align(1)
                                                    ^~~~~~~~~~~~~"

Please find the attached log of build break. Please do let me know if there is some compatiblity issue in my environment. 

Thanks in advance for the help.

xiewenxiang, 2019-07-04:
It should be because the IDF version is not correct, or submodule is not updated.

lhoyosm, 2019-07-05:
I ran into the same issue today when following the getting started instructions while installing the toolchain in a new PC.
The problem is the compiler version is not supported. The getting started link will get you the 8.2.0 version. when I replaced the xtensa-esp32-elf folder with the one in my old PC which is of version 5.2.0 the problem was solved.
You can get version 5.2.0 here: https://dl.espressif.com/dl/xtensa-esp32-elf-linux64-1.22.0-80-g6c4433a-5.2.0.tar.gz

kdpatino, 2019-07-05:
I got the same error following https://docs.espressif.com/projects/esp-idf/en/stable/get-started/index.html#get-esp-idf. 

```
In file included from /home/kenny/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/cstdlib:75,
                 from /home/kenny/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/ext/string_conversions.h:41,
                 from /home/kenny/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/bits/basic_string.h:6391,
                 from /home/kenny/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/string:52,
                 from /home/kenny/esp/esp-idf/components/asio/asio/asio/include/asio/impl/error.ipp:19,
                 from /home/kenny/esp/esp-idf/components/asio/asio/asio/include/asio/impl/src.hpp:22,
                 from /home/kenny/esp/esp-idf/components/asio/asio/asio/src/asio.cpp:11:
/home/kenny/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/stdlib.h:155:44: error: expected initializer before '__result_use_check'
 void *reallocarray(void *, size_t, size_t) __result_use_check __alloc_size(2)
                                            ^~~~~~~~~~~~~~~~~~
/home/kenny/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/stdlib.h:340:52: error: expected initializer before '__alloc_align'
 void * aligned_alloc(size_t, size_t) __malloc_like __alloc_align(1)
```

bkvenkatesh, 2019-07-05:
> It should be because the IDF version is not correct, or submodule is not updated.

bkvenkatesh, 2019-07-05:
> I ran into the same issue today when following the getting started instructions while installing the toolchain in a new linux PC.
> The problem is that the compiler version is not supported as shown at the start of the log.
> The getting started link will get you the 8.2.0 version. when I replaced the xtensa-esp32-elf folder with the one in my old PC which is of version 5.2.0 the problem was solved.
> You can get version 5.2.0 here: https://dl.espressif.com/dl/xtensa-esp32-elf-linux64-1.22.0-80-g6c4433a-5.2.0.tar.gz (linux version)
> 
> I am not sure if this is the right windows version but you could try it: https://dl.espressif.com/dl/esp32_win32_msys2_environment_and_toolchain-20181001.zip

I will try to use the older version and let you know the result. Thanks alot for letting me know about this.

mctosima, 2019-07-08:
got same issue when developping ESP-WHO to ESP-EYE
I'm using ESP-IDF that included on [ESP-WHO](https://github.com/espressif/esp-who)

`/Users/mctosima/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/stdlib.h:155:44: error: expected initializer before '__result_use_check'
 void *reallocarray(void *, size_t, size_t) __result_use_check __alloc_size(2)
                                            ^~~~~~~~~~~~~~~~~~
/Users/mctosima/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/stdlib.h:340:52: error: expected initializer before '__alloc_align'
 void * aligned_alloc(size_t, size_t) __malloc_like __alloc_align(1)
                                                    ^~~~~~~~~~~~~
[49/355] Building C object esp-face/CMakeFiles/esp-face.dir/face_detection/fd_forward.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
`

mctosima, 2019-07-08:
> got same issue when developping ESP-WHO to ESP-EYE
> I'm using ESP-IDF that included on [ESP-WHO](https://github.com/espressif/esp-who) [...]

Solve by updating xtensa

bkvenkatesh, 2019-07-08:
@mctosima : which version of the xtensa toolchain solved it? i used older version (5.2.0) to solve the build issue with esp-at

bkvenkatesh, 2019-07-08:
> I will try to use the older version and let you know the result. Thanks alot for letting me know about this.
Using the version mentioned i could get rid of the issue. Thank you

bkvenkatesh, 2019-07-11:
Using the older version of the toolchain the build break disappears. 
Thank you!

closed 2019-07-11 by bkvenkatesh

simonliu009, 2019-07-22:
Ran into the same issue today, fixed by rolling back to toolchain 5.2.0.

Mickael17, 2019-07-26:
Hi guys ! 
I have a question : what version of ESP-IDF do you use @bkvenkatesh ?
Thanks

johnny-dash, 2019-08-29:
Does anyone know where can I find the 5.2.0 version of `xtensa-esp32-elf `?

bkvenkatesh, 2019-09-27:
> Hi guys !
> I have a question : what version of ESP-IDF do you use @bkvenkatesh ?
> Thanks

> Does anyone know where can I find the 5.2.0 version of `xtensa-esp32-elf ` for mac os?

Did you find the answer for this? If so please comment for the benefit of others

xcguang, 2019-09-27:
@bkvenkatesh @johnny-dash 
You can find the download url at https://github.com/espressif/esp-at/blob/master/docs/ESP_AT_Get_Started.md

razvanphp, 2019-10-30:
> Does anyone know where can I find the 5.2.0 version of `xtensa-esp32-elf ` for mac os?

https://docs.espressif.com/projects/esp-idf/en/stable/get-started/macos-setup.html

Suraj-singh999898, 2020-03-11:
yes by using xtensa-esp32-elf 5.2.0 version problem solved but 
idf v4.0 example will stop working.

so anyone knows another way to do with xtensa 8.2.0 toolchain

Thanks for your effort

talsaiag, 2020-05-30:
> Does anyone know where can I find the 5.2.0 version of `xtensa-esp32-elf ` for mac os?

https://dl.espressif.com/dl/xtensa-esp32-elf-osx-1.22.0-80-g6c4433a-5.2.0.tar.gz
(changed linux64 -> osx)

SamuTheCoder, 2024-09-27:
If anyone is still having trouble, you can try deleting a potential previous build of the project you have on the project's directory.

Just do "rm -rf build/" and build the project again.