espressif/esp-at #216: V1.2.0 BLE server problem

zjykymt, 2019-07-04:
After changing from 1.1.3 to 1.2.0, when esp32 is used as BLE service, Android phones search for less BLE services.

- V1.2.0.0
>>> Service: 00001800-0000-1000-8000-00805f9b34fb
>>> Character: BleGattCharacter{uuid=00002aa6-0000-1000-8000-00805f9b34fb, property=2, permissions=0, descriptors=null}
>>> Character: BleGattCharacter{uuid=00002a01-0000-1000-8000-00805f9b34fb, property=2, permissions=0, descriptors=null}
>>> Character: BleGattCharacter{uuid=00002a00-0000-1000-8000-00805f9b34fb, property=2, permissions=0, descriptors=null}

>>> Service: 00001801-0000-1000-8000-00805f9b34fb
>>> Character: BleGattCharacter{uuid=00002a05-0000-1000-8000-00805f9b34fb, property=32, permissions=0, descriptors=[BleGattDescriptor{mUuid=00002902-0000-1000-8000-00805f9b34fb, mPermissions=0, mValue=null}]}

- V1.1.3.0
>>> Service: 00001800-0000-1000-8000-00805f9b34fb
>>> Character: BleGattCharacter{uuid=00002aa6-0000-1000-8000-00805f9b34fb, property=2, permissions=0, descriptors=null}
>>> Character: BleGattCharacter{uuid=00002a01-0000-1000-8000-00805f9b34fb, property=2, permissions=0, descriptors=null}
>>> Character: BleGattCharacter{uuid=00002a00-0000-1000-8000-00805f9b34fb, property=2, permissions=0, descriptors=null}

>>> Service: 00001801-0000-1000-8000-00805f9b34fb
>>> Character: BleGattCharacter{uuid=00002a05-0000-1000-8000-00805f9b34fb, property=32, permissions=0, descriptors=[BleGattDescriptor{mUuid=00002902-0000-1000-8000-00805f9b34fb, mPermissions=0, mValue=null}]}

>>> Service: 0000a002-0000-1000-8000-00805f9b34fb
>>> Character: BleGattCharacter{uuid=0000c302-0000-1000-8000-00805f9b34fb, property=8, permissions=0, descriptors=[BleGattDescriptor{mUuid=00002901-0000-1000-8000-00805f9b34fb, mPermissions=0, mValue=null}]}
>>> Character: BleGattCharacter{uuid=0000c307-0000-1000-8000-00805f9b34fb, property=2, permissions=0, descriptors=[BleGattDescriptor{mUuid=00002901-0000-1000-8000-00805f9b34fb, mPermissions=0, mValue=null}]}
>>> Character: BleGattCharacter{uuid=0000c306-0000-1000-8000-00805f9b34fb, property=32, permissions=0, descriptors=[BleGattDescriptor{mUuid=00002902-0000-1000-8000-00805f9b34fb, mPermissions=0, mValue=null}]}
>>> Character: BleGattCharacter{uuid=0000c304-0000-1000-8000-00805f9b34fb, property=8, permissions=0, descriptors=null}
>>> Character: BleGattCharacter{uuid=0000c305-0000-1000-8000-00805f9b34fb, property=16, permissions=0, descriptors=[BleGattDescriptor{mUuid=00002902-0000-1000-8000-00805f9b34fb, mPermissions=0, mValue=null}]}
>>> Character: BleGattCharacter{uuid=0000c301-0000-1000-8000-00805f9b34fb, property=2, permissions=0, descriptors=[BleGattDescriptor{mUuid=00002901-0000-1000-8000-00805f9b34fb, mPermissions=0, mValue=null}]}
>>> Character: BleGattCharacter{uuid=0000c303-0000-1000-8000-00805f9b34fb, property=4, permissions=0, descriptors=[BleGattDescriptor{mUuid=00002901-0000-1000-8000-00805f9b34fb, mPermissions=0, mValue=null}]}
>>> Character: BleGattCharacter{uuid=0000c300-0000-1000-8000-00805f9b34fb, property=2, permissions=0, descriptors=[BleGattDescriptor{mUuid=00002901-0000-1000-8000-00805f9b34fb, mPermissions=0, mValue=null}]}

xiewenxiang, 2019-07-04:
Do you create and start the services?

zjykymt, 2019-07-04:
Sorry, I can get it after I restart my phone.

I found that some Bluetooth 4.0 phones can get *Advertising Dat*a, but can't connect with ESP32, such as Galaxy s5. SM-g9006v( Andriod 5.0 ).

Can you provide some explanation for ESP32's BLE compatibility?

xiewenxiang, 2019-07-04:
Could you provide the AT log when the connection fails？

zjykymt, 2019-07-05:
There is no AT log of UART1 and UART0, ESP32-WROOM-02D does not have any command output for this BLE connection.

It is found that only this mobile phone(Galaxy s5. SM-g9006v, Andriod 5.0, Bluetooth 4.0) has this situation.

xiewenxiang, 2019-11-12:
Please try these steps
```c
AT+BLEINIT=2

OK
AT+BLEADVDATA="0201061AFF4C000215FDA50693A4E24FB1AFCFC6EB0764782527B7F206C5"

OK
AT+BLEADVSTART

OK
```

closed 2020-03-16 by xcguang