espressif/esp-at #218: Perform OTA update from v1.1.3.0 to 1.2.0.0

carlessole, 2019-07-10:
Hello.
When compiling ESP32-AT project, in the build directory there are several *.bin files that python script flash to the ESP32.

For v1.1.3.0
```
0x1000 		/build/bootloader/bootloader.bin 
0x20000 	/build/at_customize.bin 
0x21000 	/build/customized_partitions/ble_data.bin 
0x24000 	/build/customized_partitions/server_cert.bin 
0x26000 	/build/customized_partitions/server_key.bin 
0x28000 	/build/customized_partitions/server_ca.bin 
0x2a000 	/build/customized_partitions/client_cert.bin 
0x2c000 	/build/customized_partitions/client_key.bin 
0x2e000 	/build/customized_partitions/client_ca.bin 
0x30000 	/build/customized_partitions/factory_param.bin 
0xf000 		/build/phy_init_data.bin 
0x100000 	/build/esp-at.bin 
0x8000 		/build/partitions_at.bin
```

For v1.2.0.0
```
0x10000 	/build/ota_data_initial.bin 
0x1000 		/build/bootloader/bootloader.bin 
0x20000 	/build/at_customize.bin 
0x21000 	/build/customized_partitions/ble_data.bin 
0x24000 	/build/customized_partitions/server_cert.bin 
0x26000 	/build/customized_partitions/server_key.bin 
0x28000 	/build/customized_partitions/server_ca.bin 
0x2a000 	/build/customized_partitions/client_cert.bin 
0x2c000 	/build/customized_partitions/client_key.bin 
0x2e000 	/build/customized_partitions/client_ca.bin 
0x30000 	/build/customized_partitions/factory_param.bin 
0xf000 		/build/phy_init_data.bin 
0x100000 	/build/esp-at.bin 
0x8000 		/build/partitions_at.bin
```

I understand that when performing and OTA update just the `esp-at.bin` is written to the OTA partition. So are all the other .bin "**immutables**"? Is there a way to modify the other .bin files _Over-The-Air_?

And also, what about the "ota_data_initial.bin" that appears in the **v1.2.0.0**.
So, is it possible to update from v1.1.3.0 to 1.2.0.0? Or as "ota_data_initial.bin" is missing some error may occur?

Thanks.

xiewenxiang, 2019-07-11:
It can be update from v1.1.3.0 to 1.2.0.0.

carlessole, 2019-07-11:
@xiewenxiang 
Ok. 
And for understanding the other questions.

> I understand that when performing and OTA update just the esp-at.bin is written to the OTA partition. So are all the other .bin "immutables"? Is there a way to modify the other .bin files Over-The-Air?

Thanks

carlessole, 2019-07-26:
Hello again, 
What about that?

> I understand that when performing and OTA update just the esp-at.bin is written to the OTA partition. So are all the other .bin "immutables"? Is there a way to modify the other .bin files Over-The-Air?

Thank you!

SONGjiaxiu, 2019-07-29:
Hi,
you're right, all the other .bin "immutables" in the ESP32-AT project .The official AT project does not support upgrading other .bin files, However, you can upgrade other files by implementing this feature yourself.
Good luck.

carlessole, 2019-07-29:
Thanks @SONGjiaxiu,
And how you can implement that?
Because reading [Over The Air Updates](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/ota.html) of the ESP-IDF, I just see how to update the application, not the others .bin.
Thank you very much

xcguang, 2019-09-05:
OTA just a TCP connection, you can create a TCP to connect your server, and get your bin which you want to OTA, then write the data into the partition.

carlessole, 2019-09-05:
@xcguang thank for your answer.
But how it can be done.?
Could you provide a code snippet please?
Using functions that OTA updates provides: https://docs.espressif.com/projects/esp-idf/en/v3.2/api-reference/system/ota.html, just refers to the application partition I understand.
So If I want to upgrade my  "at_customize.bin", I don't know how to do via OTA update.
Or it can be done using "esp_err_t esp_ota_begin(const esp_partition_t *partition, size_t image_size, esp_ota_handle_t *out_handle)" setting *partition = 0x20000 (at_customize.bin) for example?
Thanks!

carlessole, 2019-09-19:
Anyone know how to resolve this issue?

xcguang, 2019-09-19:
You can download your "at_customize.bin" from your cloud server, referring to https://github.com/espressif/esp-at/blob/master/main/at_upgrade.c, then write the data into at_customize partition.
Don't use OTA APIs, because the binary is not an app firmware .

carlessole, 2019-09-19:
@xcguang,
Ok I understand and I know how to get the binary from the cloud server.
The probles is that I don't know how to do that:
`Don't use OTA APIs, because the binary is not an app firmware .`
I don't know how to write the partition without using the OTA api. It has to be done using plain C I guess right? And so, there is no way to verify the image eihter?
Thanks very much

xcguang, 2019-09-19:
You can call `esp_partition_find_first` to get at_customize partition information, then write data into the partition by `esp_partition_write`. 

BTW, you should erase the partition by `esp_partition_erase_range` before writing it.

carlessole, 2019-09-19:
Great, thanks!!

closed 2019-11-06 by xcguang