espressif/esp-at #220: Connected flags?

jitenshap, 2019-07-18:
I trying write custom AT Wi-Fi command.
When disabling AT wifi command support and using my AT Wi-Fi command, net commands does'nt works.

Does the net command need some Wi-Fi connected flags?

SONGjiaxiu, 2019-07-26:
Hi,
The net command is based on AT Wi-Fi command in the common AT project,  if you want to use net command, you must  enable AT wifi command. 
You can enable wifi command, and just implement your wifi commands, then send your wifi command. BTW, please donot regist callback about wifi event.

jitenshap, 2019-07-26:
Thank you for response.
I must write also net command for power saving.
When I regist a Wi-Fi command, ESP32 uses 100 mA even if CWMODE = 0.

closed 2019-07-26 by jitenshap