espressif/esp-at #222: UART0 not available for AT Command

tompie69, 2019-07-25:
Documentation states that:
_• Users can change to use another UART for AT communication. For example, if you want to use
UART0 for AT communication, you need to:
- make menuconfig -> component config -> AT -> "AT UART settings" to set it to use UART 0
- The debug log will output through UART0 by default, but users can disable it in menuconfig, as:
make menuconfig --> Component config --> ESP32-specific --> UART for console output

This is option is not available when I run make menuconfig?

SONGjiaxiu, 2019-07-30:
Hi,
The document is out of date, and the new document will be updated. Users can not change to use another UART for AT communication now. 
ths

swouf, 2019-07-31:
Hello,
I am also trying to configure the AT firmware to communicate AT commands on UART0. I saw that it is possible to change the TX, RX, CTS and RTS pin using factory parameters. Therefore I tried to set the pin for UART0 (TX=1, RX=3, CTS=19 and RTS=22) but I had no success initiating the communication on UART0 yet. Is it now the right way to do this as the option is no longer available in the menuconfig ?

Thank you in advance for your answer.

xiewenxiang, 2019-07-31:
You can directly change the source code and recompile the at project.

PATH: `/esp-at/main/interface/uart/at_uart_task.c`
FUNCTION: `static void at_uart_init(void)`  `uart_set_pin(esp_at_uart_port, tx_pin, rx_pin, rts_pin, cts_pin);`

swouf, 2019-08-01:
Thank you, this method works. I changed the defines at the beginning of the file and everything worked fine. Nevertheless, if this is the way to proceed, what is the point of having factory parameters encoded in a separate partition ? How could we use them to make these settings and eventually avoiding to rebuild and flash the whole firmware to change only this ?

This feature would be very useful, especially for production setup and updates.

xiewenxiang, 2019-08-01:
@swouf 

components/customized_partitions/raw_data/factory_param/factory_param_data.csv

xiewenxiang, 2019-08-01:
In addition, you need to modify the makefile

xiewenxiang, 2019-08-01:
[image]

closed 2019-09-05 by xcguang