espressif/esp-at #223: please port AT+CIPRECVMODE=1 from esp8266 AT 1.7.0 to this AT firmware

jandrassy, 2019-07-25:
I can use esp8266 AT firmware 1.7.0 with 500000 baud without UART flow control with passive mode (AT+CIPRECVMODE=1). If this should be esp8266's next AT firmware, will this AT firmware have passive receive mode?

I made a new WiFiEsp Arduino library for AT+CIPRECVMODE=1

SONGjiaxiu, 2019-07-26:
yeah, there are still some bugs to be fixed now, esp8266's next AT firmware will be released.

jandrassy, 2019-07-26:
> 
> 
> yeah, there are still some bugs about `AT+CIPRECVMODE=1` now, and this feature will be released in the next AT firmware.

perfect, thank you

xiewenxiang, 2019-07-31:
@jandrassy 

Please use the esp-at latest master branch, replace the lib and recompile the esp-at project.

[libat_core_for_esp8266.a.zip](https://github.com/espressif/esp32-at/files/3451090/libat_core_for_esp8266.a.zip)
[libat_core_for_esp32.a.zip](https://github.com/espressif/esp32-at/files/3451091/libat_core_for_esp32.a.zip)

jandrassy, 2019-08-03:
I tested it. 

The `+CIPRECVDATA=` response shouldn't have link id. It returns `+CIPRECVDATA=<link_id>,<len>,<data>` but the reference for AT 1.7 has `AT+CIPRECVDATA=<len>,<data>`

Other passive mode difference is that the `AT+CIPRECVLEN?`query returns -1 for inactive links. On AT 1.7 it has 0 for inactive link. But actually the -1 indication is a better option and the reference for 1.7 commands doesn't mention the value for inactive link.

The 2.0 feels older then 1.7. It has slow reactions to commands. (different priorities?). It doesn't have _CUR versions of commands.

One more difference: the CIPSEND prompt > has a space after it on 1.7, but not on 2.0

jandrassy, 2019-08-06:
in passive receive mode `+IPD,<linkId>,<len>` should be terminated by `\r\n` not by `:`

xcguang, 2019-09-05:
Please try again using the latest project.

jandrassy, 2019-09-05:
all problems are still there

jandrassy, 2019-10-26:
you made some changes for passive receive mode
fixed is `+IPD,<linkId>,<len>` termination (\r\n)

the +CIPRECVDATA has now 3 different versions:
1) documented for AT 1.7 is `+CIPRECVDATA:<actual_len>,<data>`
2) realized in AT 1.7 is `+CIPRECVDATA,<actual_len>:<data>` similar to `+IPD,<len>[,<remote	
IP>,<remote port>]:<data>`
3) currently in this AT firmware `+CIPRECVDATA:<link_id>,<actual_len>,<data>`

it is much simpler to parse +CIPRECVDATA if `:`separates meta information from <data>, so I would prefer the `+CIPRECVDATA,<actual_len>:<data>` as is now implemented in AT 1.7.x. It is a direct response to `AT+CIPRECVDATA=<link_id>,<len>` so <link_id> in response is redundant.

jandrassy, 2019-11-06:
new test after recent commits
`+CIPRECVDATA` without <link_id>, but with : before <actual_len>, so not like +IPD even if it now respects `AT+CIPDINFO=1` now

UDP still **doesn't work at all** if passive receive mode is activated. The +IPD returns only the size and AT+CIPRECVDATA returns ERROR

I still can't make SSL connection work. Is SSL still ESP32 only?

xcguang, 2019-11-07:
Hi, 
After internal discussion, we decide to keep current `+CIPRECVDATA` response format in RTOS AT.
There is some difference between Nonos AT and RTOS AT, so you cannot be completely compatible.
The other issues, we will try to reproduce them.

jandrassy, 2019-11-07:
> 
> 
> Hi,
> After internal discussion, we decide to keep current `+CIPRECVDATA` response format in RTOS AT.
> There is some difference between Nonos AT and RTOS AT, so you cannot be completely compatible.
> The other issues, we will try to reproduce them.

OK, thank you for the answer. I adapt the library code. As we progress with this, it has less and less #ifdefs for AT 2.0, but some will remain.

jandrassy, 2019-11-09:
on ESP32 SSL works OK in passive receive mode

jandrassy, 2019-11-16:
I made SSL connect on esp8266. Problem was first a missing python import xlrd and then SSLCCONF set to mode 2, which doesn't work with the demo certificates. ESP32 has it at default 0 - no authorization.

The firmware fails to report the correct length of available data on SSL connection, but it seems that the firmware now sends CLOSED only after all data are read. Is it so?

On esp8266 a SSL download of a 40 kB HTML file is unable to finish. The firmware reports with +IPD 34 bytes or less after most of +CIPRECVDATA. And sometimes suddenly the firmware sends the data with +IPD.

jandrassy, 2019-12-05:
with wolfssl it is not better. it still suddenly sends the data with +IPD.

jandrassy, 2019-12-07:
the attached capture from TX pin of the esp8266 is a download of a 40 kB web page over SSL link in passive receive mode (command echo is on and the timestamps are added by the terminal). the requested URL is `GET /app/stranky/zoznam.html HTTP/1.1`. 

it is tested with current HEAD with wolfssl. the UART is at 500000 baud and 3 kB can be returned with every CIPRECVDATA

in log you can see that the last +IPD contains data as in normal mode. this should not happen for passive receive mode.

many times the count of bytes returned with CIPRECVDATA stays on subsequent calls on the same small size like 3, 14, 21 or 177. why? problems with heap allocation?

[ssl_download_log.txt](https://github.com/espressif/esp-at/files/3935160/ssl_download_log.txt)
[sdkconfig.txt](https://github.com/espressif/esp-at/files/3935161/sdkconfig.txt)

tsvehagen, 2019-12-18:
Any updates on this? I tried using passive mode in esp-at v2.0 and when receiving on UDP link it just gives `+IPD,0,97:`. No data follows, I guess that's what you experience as well @jandrassy.

jandrassy, 2019-12-22:
> Any updates on this? I tried using passive mode in esp-at v2.0 and when receiving on UDP link it just gives `+IPD,0,97:`. No data follows, I guess that's what you experience as well @jandrassy.

the +IDP is without `:` as it would be if passive mode was supported for UDP. but AT+CIPRECVDATA for the link returns ERROR

tsvehagen, 2020-01-10:
@xcguang Is there any work being done on this? I think CIPRECVMODE=1 is a quite important feature as it's not really possible to get large about of data with TCP over a slow UART unless you can control the receive window.

jandrassy, 2020-01-10:
> 
> 
> @xcguang Is there any work being done on this? I think CIPRECVMODE=1 is a quite important feature as it's not really possible to get large about of data with TCP over a slow UART unless you can control the receive window. And you usually also need to use UDP simultaneously for DNS etc.

you could open a new issue specially for UDP with passive mode. And I will track the esp8266 SSL problems here

tsvehagen, 2020-01-10:
> > @xcguang Is there any work being done on this? I think CIPRECVMODE=1 is a quite important feature as it's not really possible to get large about of data with TCP over a slow UART unless you can control the receive window. And you usually also need to use UDP simultaneously for DNS etc.
> 
> you could open a new issue specially for UDP with passive mode. it is already in a release (ESP32 2.0.0) And I will track the esp8266 SSL problems in passive mode here

Ah ok sure. So the TCP passive mode is working as expected for you now at least? I mean the "only" problem is UDP?

jandrassy, 2020-01-10:
> Ah ok sure. So the TCP passive mode is working as expected for you now at least? I mean the "only" problem is UDP?

yes the esp32 version is good for TCP and SSL. the esp8266 has problems with SSL.

jandrassy, 2020-01-18:
with new version of lib_at_core the UDP works passive. thank you.

there is a new issue https://github.com/espressif/esp-at/issues/328, which I discovered searching for I way to get the IP address of the UDP message sender.

the SSL on esp8266 has still the same problems as with previous lib_at_core.

xcguang, 2020-03-15:
May I close this issue? If you have other question, please create a new issue.

jandrassy, 2020-03-15:
> May I close this issue? If you have other question, please create a new issue.

it is not ready. SSL on esp8266 suddenly returns data with +IDP.
and UDP passive receive requires to read the whole message with one AT+CIPRECVDATA.

jandrassy, 2021-07-10:
> > May I close this issue? If you have other question, please create a new issue.
> 
> it is not ready. SSL on esp8266 suddenly returns data with +IDP.
> and UDP passive receive requires to read the whole message with one AT+CIPRECVDATA.

these two problems are still not solved in 2.2

jandrassy, 2022-01-06:
the unexpected +IDP with data in passive mode for SSL on esp8266 is solved in the esp8266 2.2.0 branch

reading UDP message with a smaller buffer stil doesn't work https://github.com/espressif/esp-at/issues/451

closed 2022-01-06 by jandrassy