espressif/esp-at #225: AT+BLESCAN result doesn't have scan response data.

boyhuesd, 2019-08-10:
Hi all,

In [AT+BLESCAN](https://github.com/espressif/esp32-at/blob/master/docs/ESP_AT_Commands_Set.md#55-esp32-only-atblescanenables-ble-scanning) spec , it says the result will comes up with scan response data. But we can't find it in our result.

```
AT+BLESCAN=1,1

OK
+BLESCAN:c0:45:ff:00:11:3f,-67,0201020302f0ff1106f2c3f0aea9fa158c9d49ae73710a81e7020a04,,0
+BLESCAN:9c:ff:ff:cd:7e:ff,-89,1aff4c00021550765cb7d9ea4e2199a4fa879613a492a3e7c88bce,,0
+BLESCAN:9c:ff:ff:cd:7e:ff,-87,1dff2d010200011076cd75ebd467426aa66a0eaa127b8d71f3a73afe2e03,,0
```

One of our test device has its name in scan response. So if we can't get the package, there is no way to know the name of the device.

We did test on AT Firmware V1.1.3 and V1.2.0, both have same result.

Please support. Thank you all.

Best regards,
Dat Le.

xiewenxiang, 2019-09-18:
Do you set the scan params? If you set the passive scan mode, you will don't get scan RSP data.

boyhuesd, 2019-09-24:
> 
> 
> Do you set the scan params? If you set the passive scan mode, you will don't get scan RSP data.

Thank you. It's very helpful. We can get the scan response after set it to active scan.

closed 2019-09-24 by boyhuesd