espressif/esp-at #228: Assert when connecting WIFI

Neslon07, 2019-09-04:
The WROOM-32 reboots every time I try to connect to a WIFI network. 
On Debug UART I see the following log:

_PATH_TO_ESP/esp32-at/esp-idf/components/freertos/event_groups.c:350 (xEventGroupWaitBits)- assert failed!
abort() was called at PC 0x4009112c on core 1

ELF file SHA256: e50f818b577caf5742800049d6741e310c2ec3f7f505d39b68decc6bd412f886

Backtrace: 0x4008d3cf:0x3ffdf5c0 0x4008d68d:0x3ffdf5e0 0x4009112c:0x3ffdf600 0x4012fb05:0x3ffdf630 0x4012fb72:0x3ffdf700 0x4012729d:0x3ffdf720 0x40130542:0x3ffdf740

Rebooting..._

Here the version:
AT+GMR:
AT version:2.0.0.0-dev(a0e0e59 - Sep  3 2019 12:47:42)
SDK version:v3.3-beta3-170-g91f29bef1-dirty
compile time:Sep  4 2019 15:22:46

Here the steps I do:
AT+CWMODE=1
->OK
AT+CWJAP="MY ACCESS POINT","MY PASSWORD"
->ready -> Here the assert failed / reboot starts

xcguang, 2019-09-04:
Yes, I introduced a bug from commit 9549309. Please use the project before this commit id.

xcguang, 2019-09-05:
5acffca has fixed it, please try again

Neslon07, 2019-09-05:
Wifi connection works now

closed 2019-09-05 by Neslon07