espressif/esp-at #229: SSL Problems (Sending data and opening with Ethernet Enabled)

Neslon07, 2019-09-05:
Problem -> Open a SSL connection with Ethernet commands enabled (in menuconf->AT ethernet support). The SSL connection is with server CA cert and client cert and key.
   -> Open the SSL connection with ethernet cable pluged in (send data error):
        AT+CIPMUX=1   ->OK
        AT+CIPSSLCCONF=0,3,0,0 -> OK
        AT+CIPSTART=0,"SSL","serverUrl",443,180 -> OK -> 0, CONNECT
        AT+CIPSEND=0, 10 (send data after > etc.) -> ERROR (Monitor console: "link is not valid")

   ->Now open the SSL connection when WIFI is connected (no ethernet cable pluged in, open SSL port problem):
        AT+CIPMUX=1   ->OK
        AT+CIPSSLCCONF=0,3,0,0 -> OK
        AT+CIPSTART=0,"SSL","serverUrl",443,180 -> ERROR -> (Monitor console: 
               cert_len=1220
               prvtkey_len=1675
               ca_len=1188
               ctx ok
               ssl connect fail
               create 0 socket fail)

So the connection works over Ethernet but not over WIFI (with same certificates, URLS etc.) and sending data is not working at all.

If I disable in Menuconfig AT ethernet support, The SSL connection works over WIFI, but sending gives always errors as above with Ethernet:
        AT+CIPMUX=1   ->OK
        AT+CIPSSLCCONF=0,3,0,0 -> OK
        AT+CIPSTART=0,"SSL","serverUrl",443,180 -> OK -> 0, CONNECT
        AT+CIPSEND=0, 10 (send data after > etc.) -> ERROR (Monitor console "link is not valid")

Since I need Ethernet and WIFI in my project with SSL connection I hope you can fix this two issues (SSL connect on both, WIFI or Ethernet, and sending and receiving data on both). 

Note: I don't have two SSL connection open at the same time (which is not possible from documentation), so if someone plugs in ethernet while connected over WIFI, I disconnect WIFI and reconnect socket over Ethernet again.

Thank you in advance for helping me.

ubichsel, 2019-09-10:
I have exactly the same problem. Can't send data over a TLS connection:
AT+CIPMUX=1 ->OK
AT+CIPSSLCCONF=0,3,0,0 -> OK
AT+CIPSTART=0,"SSL","serverUrl",443,180 -> OK -> 0, CONNECT
AT+CIPSEND=0, 10 (send data after > etc.) -> ERROR (Monitor console: "link is not valid")

Please fix this (before adding all new http and mqtt commands), since a working TLS connection is essential.

Neslon07, 2019-10-09:
With current AT-Lib it works, thanks for fixing it.

closed 2019-10-09 by Neslon07