espressif/esp-at #230: make flash error  ： IOError:

stevenzsjcyzz, 2019-09-06:
App "esp-at" version: ESP-AT
Generating esp32.project.ld
Traceback (most recent call last):
  File "C:/msys32/home/Administrator/esp/esp-idf/tools/ldgen/ldgen.py", line 132, in <module>
    main()
  File "C:/msys32/home/Administrator/esp/esp-idf/tools/ldgen/ldgen.py", line 96, in main
    sdkconfig = SDKConfig(kconfig_file, config_file, args.env)
  File "C:/msys32/home/Administrator/esp/esp-idf/tools/ldgen/sdkconfig.py", line 56, in __init__
    self.config = kconfiglib.Kconfig(kconfig_file)
  File "C:/msys32/home/Administrator/esp/esp-idf/tools/kconfig_new/kconfiglib.py", line 656, in __init__
    self.top_node)  # prev_node
  File "C:/msys32/home/Administrator/esp/esp-idf/tools/kconfig_new/kconfiglib.py", line 1583, in _parse_block
    self._enter_file(sourced_file)
  File "C:/msys32/home/Administrator/esp/esp-idf/tools/kconfig_new/kconfiglib.py", line 1167, in _enter_file
    .format(self._filename, self._linenr, e.message))
IOError: C:/msys32/home/Administrator/esp/esp-idf/Kconfig:65: Could not open 'C:msys32homeAdministratorespesp-idfcomponentsapp_updateKconfig.projbuild' (ENOENT: No such file or directory). Perhaps the $srctree environment variable (which was unset) is set incorrectly. Note that the current value of $srctree is saved when the Kconfig instance is created (for consistency and to cleanly separate instances). Also note that e.g. $FOO in a 'source' statement does not refer to the environment variable FOO, but rather to the Kconfig Symbol FOO (which would commonly have 'option env="FOO"' in its definition).
make: *** [/home/Administrator/esp/esp-idf/components/esp32/Makefile.projbuild:51：/home/Administrator/esp/esp-idf/examples/esp-at/build/esp32/esp32.project.ld] 错误 1

Neslon07, 2019-09-06:
I had the same probleme, I think the cygpath function removes all backslashes in the path (on Windows).  I used an older ldgen.mk file and then it worked (took me a while to figure it out...).

So just replace the file in esp-idf\ldgen.mk with the one I attached. Hopefully it will work after that.
[ldgen.zip](https://github.com/espressif/esp-at/files/3582323/ldgen.zip)

xcguang, 2019-09-06:
Hi, 
Now there is a bug in the ldgen.mk, you can change "cygpath -w" to "cygpath -m" for temporary correction.
@Neslon07 It also took me some time to debug it. I have report esp-idf this bug.

stevenzsjcyzz, 2019-09-06:
@Neslon07  Thanks for your help . I will try.

stevenzsjcyzz, 2019-09-06:
@xcguang  Thanks .And esp-idf'version if need to match the esp-at'version . I suggest can upload a match version  link of esp-idf for windows and make sure it can run.

xcguang, 2019-09-06:
The idf version or ldgen.mk version? You can find IDF version at the https://github.com/espressif/esp-at/blob/master/module_config/module_esp32_default/IDF_VERSION.

stevenzsjcyzz, 2019-09-06:
@xcguang  Thanks But Ican't find whitch version can match the esp-at..... :(

xcguang, 2019-09-06:
You can check `module_config/module_esp32_default/IDF_VERSION` in you local esp-at.

stevenzsjcyzz, 2019-09-06:
@xcguang  OK. Thanks

stevenzsjcyzz, 2019-09-06:
@xcguang  Sorry . E... how to check the idf version in my local...

xcguang, 2019-09-06:
Which commit id of esp-at do you use？

stevenzsjcyzz, 2019-09-07:
@xcguang I git clone esp-idf-v3.3 in new path，and change the  $IDF_PATH  to the new path .But The shell will mkdir a new C:\msys32homeAdministratorespesp-idf  Folder to run and appear the same error.

xcguang, 2019-09-09:
@stevenzsjcyzz 
OK,we will try it. You can unset $IDF_PATH, and get the the project as following:
```
git clone --recursive https://github.com/espressif/esp-at.git
cd esp-at
make defconfig. // it will auto download the esp-idf
make
```

jscastonguay, 2019-10-03:
Hello,

@xcguang @stevenzsjcyzz 

Now the construction of esp32-at works for you? Have you found the solution of the problem of path without "/"?

Thanks

xcguang, 2019-11-13:
Same as https://github.com/espressif/esp-at/issues/232

xcguang, 2020-01-10:
Please try it using the latest master branch, if the issue persists, please reopen it again.

closed 2020-01-10 by xcguang

Emiel84, 2020-01-23:
> I had the same probleme, I think the cygpath function removes all backslashes in the path (on Windows). I used an older ldgen.mk file and then it worked (took me a while to figure it out...).
>
> So [...]

This worked for me! Thanks for this file.