espressif/esp-at #231: esp32-at with esp-idf 4.0 and websocket

boriss2oo6, 2019-09-14:
Hello,

It's a while since ESP-IDF 4.0 was released, are there any plans for a new release on ESP-AT (i did a try but i didn't succeeded to compile) ? Also 4.0 comes with WebSocket, so any plans to include this feature soon ?

Thanks in advance.

xiewenxiang, 2019-09-18:
The next release of AT will be based on IDF v3.3.

MaJerle, 2019-10-05:
@xiewenxiang when do you target next release of AT firmware for both, 32 and 8266?

xcguang, 2019-10-08:
If everything goes well, ESP32 is about this month, and ESP8266 next month.

kzoltaan, 2019-11-30:
is any news about ESP-AT with esp-idf 4.0?

xcguang, 2020-01-09:
The latest AT project supports ESP-IDF 4.0, I will close this issue, if you have any question, please create the new issue, thanks.

closed 2020-01-09 by xcguang