espressif/esp-at #232: menuconfig failed

jscastonguay, 2019-09-17:
Hello,

Firstly, I am new on ESP32 world.

I need an esp-at version with classic BT enable. So I tried to build to esp-at application with the latest ESP-IDF. The ESP-IDF Tools Installer used was “https://dl.espressif.com/dl/esp-idf-tools-setup-2.0.exe”. When I did “idf.py menuconfig”, the following error occurred:

“CMakeLists.txt not found in project directory …”

I suspect that I do not use the good IDF version. Am I right? Did I miss something?

Thanks,

JS

xiewenxiang, 2019-09-18:
Cmake is not currently supported in the esp-at project. Please compile with make.

jscastonguay, 2019-09-24:
Hello,

Thanks for your reply. So if I understand correctly, I cannot use idf-esp latest version which uses CMake (called by idf.py I guess when I did “idf.py menuconfig”).

I tried with idf-esp stable version 3.3 which use make (make menuconfig and so). However, I got the following error message:

> Makefile:28: There is no module_config/module_wroom-32,Using module_config/module_esp32_default
> /home/Proprietaire/esp32-at/esp32-at/tools/mkfatfs/Makefile.projbuild:13: avertissement : surchargement de la recette pour la cible « clean »
> /home/Proprietaire/esp/esp-idf/components/app_update/Makefile.projbuild:48: avertissement : ancienne recette ignorée pour la cible « clean »
> Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
> Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
> Compiler version: 5.2.0
> Python requirements from C:/msys32/home/Proprietaire/esp/esp-idf/requirements.txt are satisfied.
> 
> App "esp-at" version: ESP-AT
> Generating esp32.project.ld
> Traceback (most recent call last):
>   File "C:/msys32/home/Proprietaire/esp/esp-idf/tools/ldgen/ldgen.py", line 113, in <module>
>     main()
>   File "C:/msys32/home/Proprietaire/esp/esp-idf/tools/ldgen/ldgen.py", line 96, in main
>     sdkconfig = SDKConfig(kconfig_file, config_file, args.env)
>   File "C:/msys32/home/Proprietaire/esp/esp-idf/tools/ldgen/sdkconfig.py", line 56, in __init__
>     self.config = kconfiglib.Kconfig(kconfig_file.name)
>   File "C:/msys32/home/Proprietaire/esp/esp-idf/tools/kconfig_new/kconfiglib.py", line 656, in __init__
>     self.top_node)  # prev_node
>   File "C:/msys32/home/Proprietaire/esp/esp-idf/tools/kconfig_new/kconfiglib.py", line 1583, in _parse_block
>     self._enter_file(sourced_file)
>   File "C:/msys32/home/Proprietaire/esp/esp-idf/tools/kconfig_new/kconfiglib.py", line 1167, in _enter_file
>     .format(self._filename, self._linenr, e.message))
> IOError: C:/msys32/home/Proprietaire/esp/esp-idf/Kconfig:59: Could not open 'C:msys32homeProprietaireespesp-idfcomponentsapp_updateKconfig.projbuild' (ENOENT: No such file or directory). Perhaps the $srctree environment variable (which was unset) is set incorrectly. Note that the current value of $srctree is saved when the Kconfig instance is created (for consistency and to cleanly separate instances). Also note that e.g. $FOO in a 'source' statement does not refer to the environment variable FOO, but rather to the Kconfig Symbol FOO (which would commonly have 'option env="FOO"' in its definition).
> make: *** [/home/Proprietaire/esp/esp-idf/components/esp32/Makefile.projbuild:51: /home/Proprietaire/esp32-at/esp32-at/build/esp32/esp32.project.ld] Error 1
> 

I think that I followed the procedure correctly:
https://github.com/espressif/esp-at/blob/master/docs/ESP_AT_Get_Started.md
https://docs.espressif.com/projects/esp-idf/en/stable/get-started/index.html#get-started-get-esp-idf
https://docs.espressif.com/projects/esp-idf/en/stable/get-started/windows-setup.html

Again, I am new to ESP32. Can you please help me to build the esp32-at with classic BT enabled?

Thanks,

JS

xcguang, 2019-09-24:
Hi,
Please refer to https://github.com/espressif/esp-at/issues/230 to avoid the compile error.
Usually you needn't specify the esp-idf, it will download the correct version when compiling.
You can try as following:
1. unset IDF_PATH and rm esp-idf if it exists in the esp-at directory
2. make menuconfig(it will automatically download esp-idf), and enable `Component config` > `AT` > `AT bt command support.` 
3. Replace `cygpath -w` with `cygpath -m` as  https://github.com/espressif/esp-at/issues/230
4. make

jscastonguay, 2019-09-26:
Hello,

Thanks for your reply. It is very appreciated.

- I unset IDF_PATH.
- I did `make menuconfig` and make sure that the bt command support is enabled.
- I observed that the esp-idf is downloaded correctly.
- Then I changed the `cygpath -w` for `cygpath -m` at three locations in `ldgen.mk`. I attached an image of the `ldgen.mk` file modified.
- I did `make`.
- Same error, no “/” in path.
- Note that I tried `echo $IDF_PATH`, just to make sure and the path is `~/esp32-at/esp32-at`.

> Makefile:28: There is no module_config/module_wroom-32,Using module_config/module_esp32_default
> /home/Proprietaire/esp32-at/esp32-at/tools/mkfatfs/Makefile.projbuild:13: avertissement : surchargement de la recette pour la cible « clean »
> /home/Proprietaire/esp32-at/esp32-at/esp-idf/components/app_update/Makefile.projbuild:48: avertissement : ancienne recette ignorée pour la cible « clean »
> Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
> Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
> Compiler version: 5.2.0
> Python requirements from C:/msys32/home/Proprietaire/esp32-at/esp32-at/esp-idf/requirements.txt are satisfied.
> 
> App "esp-at" version: ESP-AT
> Generating esp32.project.ld
> Traceback (most recent call last):
>   File "C:/msys32/home/Proprietaire/esp32-at/esp32-at/esp-idf/tools/ldgen/ldgen.py", line 113, in <module>
>     main()
>   File "C:/msys32/home/Proprietaire/esp32-at/esp32-at/esp-idf/tools/ldgen/ldgen.py", line 96, in main
>     sdkconfig = SDKConfig(kconfig_file, config_file, args.env)
>   File "C:/msys32/home/Proprietaire/esp32-at/esp32-at/esp-idf/tools/ldgen/sdkconfig.py", line 56, in __init__
>     self.config = kconfiglib.Kconfig(kconfig_file.name)
>   File "C:/msys32/home/Proprietaire/esp32-at/esp32-at/esp-idf/tools/kconfig_new/kconfiglib.py", line 656, in __init__
>     self.top_node)  # prev_node
>   File "C:/msys32/home/Proprietaire/esp32-at/esp32-at/esp-idf/tools/kconfig_new/kconfiglib.py", line 1583, in _parse_block
>     self._enter_file(sourced_file)
>   File "C:/msys32/home/Proprietaire/esp32-at/esp32-at/esp-idf/tools/kconfig_new/kconfiglib.py", line 1167, in _enter_file
>     .format(self._filename, self._linenr, e.message))
> IOError: C:/msys32/home/Proprietaire/esp32-at/esp32-at/esp-idf/Kconfig:59: Could not open 'C:msys32homeProprietaireesp32-atesp32-atesp-idfcomponentsapp_updateKconfig.projbuild' (ENOENT: No such file or directory). Perhaps the $srctree environment variable (which was unset) is set incorrectly. Note that the current value of $srctree is saved when the Kconfig instance is created (for consistency and to cleanly separate instances). Also note that e.g. $FOO in a 'source' statement does not refer to the environment variable FOO, but rather to the Kconfig Symbol FOO (which would commonly have 'option env="FOO"' in its definition).
> make: *** [/home/Proprietaire/esp32-at/esp32-at/esp-idf/components/esp32/Makefile.projbuild:51: /home/Proprietaire/esp32-at/esp32-at/build/esp32/esp32.project.ld] Error 1
> 

Did I miss something? BTW Your help is very appreciated.

Thanks,

JS

[image]

MaJerle, 2019-09-28:
When can we expect CMake support in `esp-at` to be able to compile with latest ESP-IDF?

xcguang, 2019-09-28:
We will support it when esp-idf in the esp-at project is updated v4.0.  It may take a while to wait,  now esp-idf v4.0 is not released.

jscastonguay, 2019-10-01:
Hello,

Has anyone settled the problem of issue 230? I still have the problem with the path without "/" and I do not know how to build the applicaton esp32-at.

jscastonguay, 2019-10-03:
Ok it seems that I found the solution: in `ldgen.mk`, the following:

`cygpath -w $(info)` and `cygpath -w $(k)`

must be changed by

`cygpath -m '$(info)'` and `cygpath -m '$(k)'`

respectively. The single quotation is important.

reshma26p, 2019-11-08:
i have same problem but how to change cygpath -w by cygpath -m

reshma26p, 2019-11-08:
same problem by changing cygpath -w of cygpath -m 
please give solution,i need bluetooth by esp-at

xcguang, 2020-01-10:
Please try it using the latest master branch, if the issue persists, please reopen it again.

closed 2020-01-10 by xcguang