espressif/esp-at #235: AT+BLESCANPARAM

marrakchino, 2019-09-26:
What are the units for this AT command, in particular scan_window and scan_interval ? There is no information in the AT instruction set ...

Thanks!
Nabil

xiewenxiang, 2019-09-27:
`Scan interval`. 

This is defined as the time interval from when the Controller started its last LE scan until it begins the subsequent LE scan.

  - Range: 0x0004 to 0x4000 Default: 0x0010 (10 ms)
  - Time = N * 0.625 msec
  - Time Range: 2.5 msec to 10.24 seconds

`Scan window`. 

The duration of the LE scan. LE_Scan_Window shall be less than or equal to LE_Scan_Interval

  - Range: 0x0004 to 0x4000 Default: 0x0010 (10 ms)
  - Time = N * 0.625 msec
  - Time Range: 2.5 msec to 10.24 seconds

marrakchino, 2019-09-27:
Thanks @xiewenxiang .
Given the following command  :  `AT+BLESCANPARAM=0,0,0,320,40`, does this set the following parameters?
* interval : 320 * 0.625 msec = 200 msec
* window : 40 * 0.625 msec = 25 msec

marrakchino, 2019-10-03:
Anyone ?

xiewenxiang, 2019-10-08:
@marrakchino 
Yes, you are right.

closed 2019-11-12 by xcguang