espressif/esp-at #239: [AT][CIPDNS][docs] Wrong CIPDNS format on query

MaJerle, 2019-09-28:
https://github.com/espressif/esp-at/blob/master/docs/ESP_AT_Commands_Set.md#426-atcipdns--configures-domain-name-system

Docs says on query, this is response: `+CIPDNS:<enable>[,<"DNS IP1">,<"DNS IP2">,<"DNS IP3">]`.

Reality:
```
AT+CIPDNS?
+CIPDNS:8.8.8.8
+CIPDNS:8.8.4.4
OK
```

xcguang, 2019-10-04:
Please give the AT version by AT+GMR.

MaJerle, 2019-10-04:
Esp32 1.2 from espressif website.

Its very unclear how to compile this under windows by myself.

xiewenxiang, 2019-10-08:
@MaJerle 

Thank you for your information and we will fix it as soon as possible

xcguang, 2019-11-12:
Has updated.

closed 2019-11-12 by xcguang

jandrassy, 2019-11-12:
I expected the reference to change, not the implementation which was the same as on esp8266

xcguang, 2019-11-12:
Currently response is `+CIPDNS:<enable>[,<"DNS IP1">,<"DNS IP2">,<"DNS IP3">]`