espressif/esp-at #242: AT+CIPCLOSE for SSL connection and CIPMUX=1 Crashes (reboot)

Neslon07, 2019-10-14:
AT+GMR
->AT version:2.0.0.0-dev(8986ee3 - Sep 30 2019 07:33:23)
->SDK version:v3.3-dirty
->compile time:Oct  8 2019 16:22:49
->Bin version:00.10(WROOM-32)

AT+CIPCLOSE crashes the chip when closing a SSL connection (TCP / UDP works). Closing a SSL connection without multiple connection (AT+CIPMUX=0) works as well. Closing SSL connection with CIPMUX=1 crashes chip / reboots them:

AT+CIPMUX=1
->OK
AT+CIPSSLCCONF=0,0
->OK
AT+CIPSTART=0,"SSL","espressif.com",443,120
->0, Connect
->OK
AT+CIPCLOSE=0
->ready (reboot)

**Monitor Console:**
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandl
ed.
Core 1 register dump:
PC      : 0x401b6afd  PS      : 0x00060330  A0      : 0x8012fb82  A1      : 0x3f
ffb670
0x401b6afd: SSL_read at C:/msys32/home/temp/esp32-at/esp-idf/
components/openssl/library/ssl_lib.c:1550

A2      : 0x00000000  A3      : 0x3fff0538  A4      : 0x00000b68  A5      : 0x3f
ff0448
A6      : 0x3fffb728  A7      : 0x00000001  A8      : 0x00000003  A9      : 0x3f
ffb650
A10     : 0x3fff0538  A11     : 0x00000b68  A12     : 0x00060021  A13     : 0x3f
ffb728
A14     : 0x3fffb66c  A15     : 0x0000003e  SAR     : 0x00000005  EXCCAUSE: 0x00
00001d
EXCVADDR: 0x00000038  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xff
ffffff

ELF file SHA256: e90dec5758fe48fa47ff86d49a340a3542ca10a89e20d6c8aebe47595e3db73
4

Backtrace: 0x401b6afd:0x3fffb670 0x4012fb7f:0x3fffb690 0x4012fdc8:0x3fffb700 0x4
0130091:0x3fffb720
0x401b6afd: SSL_read at C:/msys32/home/temp/esp32-at/esp-idf/
components/openssl/library/ssl_lib.c:1550

0x4012fb7f: at_setupCmdCipstartEx at ??:?

0x4012fdc8: at_setupCmdCipstartEx at ??:?

0x40130091: at_setupCmdCipstartEx at ??:?

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

ustccw, 2019-10-15:
@Neslon07 thanks for your information. 
We had fixed it, please try the latest version again.

Neslon07, 2019-10-21:
I can confirm it is fixed now. Thank you!

closed 2019-10-21 by Neslon07