espressif/esp-at #243: AT+RFPOWER for Bluetooth only

Neslon07, 2019-10-14:
AT+GMR
->AT version:2.0.0.0-dev(8986ee3 - Sep 30 2019 07:33:23)
->SDK version:v3.3-dirty
->compile time:Oct 8 2019 16:22:49
->Bin version:00.10(WROOM-32)

How to set the RF-Power of BLE device if WIFI is disabled? AT+RFPOWER command gives back error but it seems the ble rf power is set correctly.

AT+CWMODE=0
->OK
AT+BLEINIT=2
->OK
AT+BLEGATTSSRVCRE
->OK
AT+BLEGATTSSRVSTART
->OK
AT+BLEADVSTART
->OK
AT+RFPOWER=82,3,3,3
->Error
From BLE analyzer I can see the rf power is set correctly but still get error code back from console.
AT+RFPOWER?
->+RFPOWER:0,3,3,3
->OK

xcguang, 2019-11-12:
Yes, you are right, this feature should be supported.

xcguang, 2020-03-12:
Can you try the latest master branch? It seems we have fixed it.

xcguang, 2020-03-29:
No update for a long time, close it.

closed 2020-03-29 by xcguang

woaijiaohaer, 2024-04-11:
hello，ESP32C3 AT+RFPOWER指令设置BLE从机的发射功率设置失败，看起来和[Neslon07](https://github.com/Neslon07)提出的issue很像，使用的是esp-at v3.2.0.0版本的代码。