espressif/esp-at #244: Changing TLS certificates on the fly

kurta999, 2019-10-15:
In future it would be very good to be able to change TLS certificates on the fly. The solution that it is "hard-coded" during flashing is very limiting. I think lot of people would use it. Is it planned in near future?

Quectel's solution looks like this and it's perfect.  (for their M95 chip)

```
AT+QSECWRITE=”RAM:ca_cert.pem”,1614,100  // Upload the CA certificate to RAM.
CONNECT
<Input the ca_cert.pem data, the size is 1614 bytes>
+QSWCWRITE: 1614,4093
OK
```

Here is their datasheet for example
[Quectel_GSM_SSL_TCP_Application_Note_V3.1.pdf](https://github.com/espressif/esp-at/files/3727659/Quectel_GSM_SSL_TCP_Application_Note_V3.1.pdf)

Similar solution would be very good for MQTT certificates and BLE characteristics too. To be able to change everything on-the-fly without need to re-flash. (When ESP being used as slave with a master MCU, it's not possible to upgrade firmware on master MCU with new encryption keys, because the master MCU can't overwrite those keys in ESP via AT)

Thanks!

xcguang, 2019-10-15:
It looks similar  to ` AT+SYSFLASH`, is it helpful you ?

kurta999, 2019-10-16:
Thank you, this is exactly what I need. Somehow I missed this function.

Is it planned to be available for ESP8266? (We used 8266 mostly)

Neslon07, 2019-10-21:
I update the certificate over the ROM-Bootloader. Generate the image from certificate with the python script (as it is done during compulation) and then update the correct flash section with ROM-Bootloader. It's not elegant and the way you describe above with load the certificate into RAM would be much nicer but it works as it is right now.

closed 2020-01-09 by xcguang