espressif/esp-at #245: Loading CA to authenticate server failed

arsickatarina, 2019-10-18:
Hi,
I'm trying to create SSL connection using ESP32-AT commands on ESP32_DevKitc_V4 board with ESP32-Solo-1 chip. I have release/v1.2.0.0 AT firmware version.
I want to authenticate server only, so i flashed certificate to 0x24000 and CA certificate to 0x28000 with ESP32 Download Tool. I converted certificates to bin format with AtPKI.py and checked if certificates are on the flash with at+sysflash=2...
My steps are:
AT+CWMODE=1
AT+CWJAP="<SSID>","<password>"
AT+CIPSNTPCFG=1,1,"1.us.pool.ntp"
AT+CIPSSLCCONF=2,0,0
AT+CIPSTART="SSL","www.google.com",443
For the last command i'm getting ERROR without any error message.
Am i missing some step?
I'd be very grateful for any kind of help.

ustccw, 2019-10-21:
@arsickatarina Hi, it looks like `AT+CIPSTART` works well on my side.

[image]

**Some tips:**
1. please try to make sure SNTP has got a valid time by `AT+CIPSNTPTIME?`
if not, certificate Validity verification will be failed.

2. please test with the following CA certificate, copy to cover `components/customized_partitions/raw_data/client_ca/client_ca_00.crt`
```
-----BEGIN CERTIFICATE-----
MIIESjCCAzKgAwIBAgINAeO0mqGNiqmBJWlQuDANBgkqhkiG9w0BAQsFADBMMSAw
HgYDVQQLExdHbG9iYWxTaWduIFJvb3QgQ0EgLSBSMjETMBEGA1UEChMKR2xvYmFs
U2lnbjETMBEGA1UEAxMKR2xvYmFsU2lnbjAeFw0xNzA2MTUwMDAwNDJaFw0yMTEy
MTUwMDAwNDJaMEIxCzAJBgNVBAYTAlVTMR4wHAYDVQQKExVHb29nbGUgVHJ1c3Qg
U2VydmljZXMxEzARBgNVBAMTCkdUUyBDQSAxTzEwggEiMA0GCSqGSIb3DQEBAQUA
A4IBDwAwggEKAoIBAQDQGM9F1IvN05zkQO9+tN1pIRvJzzyOTHW5DzEZhD2ePCnv
UA0Qk28FgICfKqC9EksC4T2fWBYk/jCfC3R3VZMdS/dN4ZKCEPZRrAzDsiKUDzRr
mBBJ5wudgzndIMYcLe/RGGFl5yODIKgjEv/SJH/UL+dEaltN11BmsK+eQmMF++Ac
xGNhr59qM/9il71I2dN8FGfcddwuaej4bXhp0LcQBbjxMcI7JP0aM3T4I+DsaxmK
FsbjzaTNC9uzpFlgOIg7rR25xoynUxv8vNmkq7zdPGHXkxWY7oG9j+JkRyBABk7X
rJfoucBZEqFJJSPk7XA0LKW0Y3z5oz2D0c1tJKwHAgMBAAGjggEzMIIBLzAOBgNV
HQ8BAf8EBAMCAYYwHQYDVR0lBBYwFAYIKwYBBQUHAwEGCCsGAQUFBwMCMBIGA1Ud
EwEB/wQIMAYBAf8CAQAwHQYDVR0OBBYEFJjR+G4Q68+b7GCfGJAboOt9Cf0rMB8G
A1UdIwQYMBaAFJviB1dnHB7AagbeWbSaLd/cGYYuMDUGCCsGAQUFBwEBBCkwJzAl
BggrBgEFBQcwAYYZaHR0cDovL29jc3AucGtpLmdvb2cvZ3NyMjAyBgNVHR8EKzAp
MCegJaAjhiFodHRwOi8vY3JsLnBraS5nb29nL2dzcjIvZ3NyMi5jcmwwPwYDVR0g
BDgwNjA0BgZngQwBAgIwKjAoBggrBgEFBQcCARYcaHR0cHM6Ly9wa2kuZ29vZy9y
ZXBvc2l0b3J5LzANBgkqhkiG9w0BAQsFAAOCAQEAGoA+Nnn78y6pRjd9XlQWNa7H
TgiZ/r3RNGkmUmYHPQq6Scti9PEajvwRT2iWTHQr02fesqOqBY2ETUwgZQ+lltoN
FvhsO9tvBCOIazpswWC9aJ9xju4tWDQH8NVU6YZZ/XteDSGU9YzJqPjY8q3MDxrz
mqepBCf5o8mw/wJ4a2G6xzUr6Fb6T8McDO22PLRL6u3M4Tzs3A2M1j6bykJYi8wW
IRdAvKLWZu/axBVbzYmqmwkm5zLSDW5nIAJbELCQCZwMH56t2Dvqofxs6BBcCFIZ
USpxu6x6td0V7SvJCCosirSmIatj/9dSSVDQibet8q/7UK4v4ZUN80atnZz1yg==
-----END CERTIFICATE-----
```

Emm, AT will add more logs for user to debug more details.

Neslon07, 2019-10-21:
I had problems with Bluetooth coexistence. There was a heap allocation problem when opening the TLS connection (and there is just a ERROR message coming back as you describe). After tuning some parameters I was able to bring it to work in parallel. Just in case you have bluetooth enabled.

katarinaar, 2019-10-25:
@ustccw This certificate only works for google... Can you please tell me what is this certificate? Why do i need to flash client CA (not server CA) in order to authenticate server?

ustccw, 2019-10-28:
> @ustccw This certificate only works for google... Can you please tell me what is this certificate? Why do i need to flash client CA (not server CA) in order to authenticate server?

The certificate is the CA certificate, you had mentioned `I want to authenticate server only`.
This CA certificate will verify server certificate.

it is unnecessary to flash `client CA` while google will allow any client to visit.

katarinaar, 2019-10-28:
@ustccw If i understend, if i want to authenticate server (in this case google) i need to flash its CA certificate to client_ca partition (0x2e000) and if i want some server to authenticate my device i need to flash mine CA certificate to server_ca partition (0x28000)?

Neslon07, 2019-10-28:
client_ca -> CA Root certificate from google to let client identify Server
client_cert + client_key ->Client certificate and key to let server identify client

katarinaar, 2019-10-28:
Thank you! I managed to authenticate server. 
But i still can't make server authenticate client.
I'm trying with https://client.badssl.com. I downloaded badssl.com-client.p12 from https://badssl.com/download/ and converted it to cert and key and then i flashed them to client_cert, client_key user partitions. But when i try to connect with AT+CIPSTART="SSL","client.badssl.com",443 i'm getting ERROR...
I'm thinking maybe reason is that ESP32 doesn't support SNI. But i couldn't find server that doesn't support SNI and requires client authentication.

Neslon07, 2019-10-28:
Do you set AT+CIPSSLCCONF=3,0,0 ?
I use it with AWS (server and client certs) and it works. There is no SNI involved but if you need port 443 you need ALPN (which is not supported from ESP-AT but from underlaying mbedtls library). I made a custom AT-Command to inject ALPN name to ssl connection. Bit dirty but it works.

marcosantonocito, 2021-02-24:
Hi @Neslon07, we are facing the same issue. It would be great if you can share the custom AT-Command to inject ALPN name to ssl connection with us! Thank you!

ustccw, 2021-02-25:
@marcosantonocito @Neslon07 About the SNI and ALPN issues, please take a look:
- https://docs.espressif.com/projects/esp-at/en/latest/AT_Command_Set/TCP-IP_AT_Commands.html#cmd-sslcsni
- https://docs.espressif.com/projects/esp-at/en/latest/AT_Command_Set/TCP-IP_AT_Commands.html#cmd-sslcalpn

closed 2021-04-14 by FayeY