espressif/esp-at #248: MQTT - blank message

Wtpaul, 2019-10-22:
Hi
I'm trying to send a blank message via MQTT to AWS
I've tried both MQTTPUB and MQTTPUBRAW, but they both return an ERROR

> AT+MQTTPUB=0,"$aws/things/Thing_Test02/shadow/get","",1,0
> ERROR

> AT+MQTTPUBRAW=0,"$aws/things/Thing_Test02/shadow/get",0,1,0
> ERROR

Any advice would be appreciated
Thanks

Wtpaul, 2019-10-22:
Just a quick update.
For AWS a blank message can also be "{}" i.e. a blank json message!
so the following has worked for me

> AT+MQTTPUBRAW=0,"$aws/things/Thing_Test02/shadow/get",2,1,0
> {}
> OK
> >+MQTTPUB:OK
> +MQTTSUBRECV:0,"$aws/things/Thing_Test02/shadow/get/accepted", .....

Sorry if this was obvious... maybe it can help someone else.

Maher9494, 2019-11-18:
Hello, Can you please provide me with the MQTT binaries as I'm having problems in generating them ?

ustccw, 2019-12-18:
@Wtpaul you could take a look at AT log port, it will print MQTT error code, please refer to: https://github.com/espressif/esp-at/blob/master/docs/ESP_AT_Commands_Set.md#MQTTErrCod, 
if there is any MQTT command return ERROR.

ustccw, 2019-12-18:
@Maher9494 could you please describe the issue in detail?

closed 2021-04-14 by FayeY