espressif/esp-at #250: SNI support

katebo, 2019-10-25:
Does ESP-AT firmware support SNI?

xcguang, 2019-10-25:
We've added it to the to-do list, but we don't have a detailed schedule. We we are waiting esp-tls component support it.

daveatpanic, 2019-10-31:
Does ESP-TLS not support SNI? It's listed here: https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/protocols/esp_tls.html

pshank, 2019-10-31:
I thought I also saw a previous issue where SNI was integrated with the library. Was I seeing things, or is SNI not supported? That would explain quite a bit if this were the case...

daveatpanic, 2019-11-01:
Nope, this says the AT firmware's not sending SNI: https://check-tls.akamaized.net/v1/tlssni.json

FayeY, 2021-04-14:
SNI seems to be supported now. https://docs.espressif.com/projects/esp-at/en/latest/AT_Command_Set/TCP-IP_AT_Commands.html#cmd-sslcsni

closed 2021-04-14 by FayeY