espressif/esp-at #252: No "+LINK_CONN" or "XX,CONNECT" message when station connects to ESP SoftAP

tvandergeer, 2019-10-28:
In previous versions of the ESP software (e.g. 1.7.0 for ESP8266) a message was sent by the ESP when a station (e.g. mobile phone) connected to the ESP's SoftAP. The [documentation on the AT+SYSMSG command](https://github.com/espressif/esp-at/blob/master/docs/ESP_AT_Commands_Set.md#cmd-SYSMSG) for v2.0 hints that there should be a similar message: 

> It will supply "+LINK_CONN : status_type,link_id,ip_type,terminal_type,remote_ip,remote_port,local_port" instead of "XX,CONNECT".

I'ver performed several tests on an ESP32 and ESP8266 on the v2.0 firmware, but the message never appeared. Is this a known issue? Is it possible to bring back a notification when a station connects to the SoftAP?

I've discussed this with @MaJerle[ in this ticket on his ESP_AT_Lib project](https://github.com/MaJerle/ESP_AT_Lib/issues/48). He confirmed that the message is no longer there.

MaJerle, 2019-10-28:
@tvandergeer `+LINK_CONN` is still available. Missing one is `+STA_CONNECTED`, `+STA_DISCONNECTED`, `+STA_GOT_IP`.

tvandergeer, 2019-10-28:
@MaJerle : You're right! I've modified my initial question accordingly to prevent confusion.

xcguang, 2019-10-29:
Yes, I will firstly add it on the ESP32 AT. There is something  is based on SDK, but current ESP8266 SDK is missing it.

xcguang, 2019-11-05:
The message on the ESP32 AT has been added.

tvandergeer, 2019-11-05:
Great, thank you!

MaJerle, 2019-11-05:
@xcguang could you please upload complete compiled project here? Thanks

xcguang, 2020-03-22:
Done on ESP8266 via 75ee09f84625aa

xcguang, 2020-03-22:
I will close this issue, if have any question, please reopen it or create a new issue.

closed 2020-03-22 by xcguang