espressif/esp-at #254: Issue receiving data from MQTT server

Masi2018, 2019-10-30:
I am using Wroom-32 as wifi module in my application . I have connected the my MCU to thw WIFi nodule through Uart port. When I program the Wroom module with old version of esp-at firmware I don't have any issues sending and receiving information to/from the MQTT server while when I use the latest code (the one is in the master branch), most of the time I can't receive data from MQTT server while I see the response has been sent from the server ( I use MQTT Explorer to monitor the data) . I don't have problem in sending data but nothing shows up on TX port of the ESP32 module which is connected to my MCU RX port.

xcguang, 2019-11-05:
Hi, 
Can you provide all command sequences starting with the first command? We will try it.

Masi2018, 2019-11-05:
@xcguang  Please see the command sequences below:

[image]

Masi2018, 2019-11-12:
@xcguang Hi, Just wanna know that if you had a chance to have a look at this issue ?

jandrassy, 2019-11-12:
did you set `AT+CIPRECVMODE=1` to use the passive recieve mode with `AT+CIPRECVDATA`?

Masi2018, 2019-11-12:
@jandrassy  No I didn't, you can see CIPRECVMODE is 0 in my first screenshot.

xcguang, 2019-11-13:
We tested it, but it looks well. Could you test it on PC serial port?

jandrassy, 2019-11-13:
> @jandrassy No I didn't, you can see CIPRECVMODE is 0 in my first screenshot.

so? you want to use it but didn't activate it

Masi2018, 2019-11-13:
@jandrassy No I don't want use this. I want the received socket data is sent instantly to host MCU through UART. So I don't want to activate this

jandrassy, 2019-11-13:
> 
> 
> @jandrassy No I don't want use this. I want the received socket data is sent instantly to host MCU through UART. So I don't want to activate this

then don't use `AT+CIPRECVLEN?'. it is part of the passive mode approach

Masi2018, 2019-11-13:
@jandrassy I only used AT+CIPRECVLEN to demonstrate that although I haven't used passive mode,  data is buffered , in my real code I don't use that. As I had mentioned my issue is after sometimes I don't receive any data from MQTT while it has been sent. Only I can get the data shortly after the connection is made, After that data is sent to the server but not received by the ESP32 and looks like it starts buffering the received data instead of sending out to the port while it is in active mode

jandrassy, 2019-11-13:
the data are always buffered. the TCP packet is read much faster from network then it can be pushed over UART. so focus on why you didn't receive `+IPD`

Masi2018, 2019-11-13:
@jandrassy  I don't have issue with older revision only in this new revision I have issue. data is not sent out to the UART port

xcguang, 2020-03-22:
Can you try the latest master branch? We did some process optimization, I'm not sure whether it affects this issue, because we cannot reproduce it.

closed 2020-08-24 by xcguang

niziak, 2020-09-01:
@Masi2018 Any results?